ThomasKaiser/sbc-bench #9: Uploading the results through a proxy

gounthar, 2019-11-12:
Hi,

I launched your script on an OrangePi RK3399 (unfortunately behind a corporate proxy), and got this error:

```bash
Full results uploaded to <!DOCTYPE html PUBLIC "-//W3C//DTD HTML 4.01//EN" "http://www.w3.org/TR/html4/strict.dtd">
<html><head>
<meta http-equiv="Content-Type" content="text/html; charset=utf-8">
<title>ERROR: The requested URL could not be retrieved</title>
<style type="text/css"><!--
 /*
 Stylesheet for Squid Error pages
 Adapted from design by Free CSS Templates
 http://www.freecsstemplates.org
 Released for free under a Creative Commons Attribution 2.5 License
*/

/* Page basics */
* {
        font-family: verdana, sans-serif;
}

html body {
        margin: 0;
        padding: 0;
        background: #efefef;
        font-size: 12px;
        color: #1e1e1e;
}

/* Page displayed title area */
#titles {
        margin-left: 15px;
        padding: 10px;
        padding-left: 100px;
        background: url('http://www.squid-cache.org/Artwork/SN.png') no-repeat left;
}

/* initial title */
#titles h1 {
        color: #000000;
}
#titles h2 {
        color: #000000;
}

/* special event: FTP success page titles */
#titles ftpsuccess {
        background-color:#00ff00;
        width:100%;
}

/* Page displayed body content area */
#content {
        padding: 10px;
        background: #ffffff;
}

/* General text */
p {
}

/* error brief description */
#error p {
}

/* some data which may have caused the problem */
#data {
}

/* the error message received from the system or other software */
#sysmsg {
}

pre {
    font-family:sans-serif;
}

/* special event: FTP / Gopher directory listing */
#dirmsg {
    font-family: courier;
    color: black;
    font-size: 10pt;
}
#dirlisting {
    margin-left: 2%;
    margin-right: 2%;
}
#dirlisting tr.entry td.icon,td.filename,td.size,td.date {
    border-bottom: groove;
}
#dirlisting td.size {
    width: 50px;
    text-align: right;
    padding-right: 5px;
}

/* horizontal lines */
hr {
        margin: 0;
}

/* page displayed footer area */
#footer {
        font-size: 9px;
        padding-left: 10px;
}

body
:lang(fa) { direction: rtl; font-size: 100%; font-family: Tahoma, Roya, sans-serif; float: right; }
:lang(he) { direction: rtl; float: right; }
 --></style>
</head><body>
<div id="titles">
<h1>ERROR</h1>
<h2>The requested URL could not be retrieved</h2>
</div>
<hr>

<div id="content">
<p><b>Invalid Request</b> error was encountered while trying to process the request:</p>

<blockquote id="data">
<pre>POST / HTTP/1.1
Host: ix.io
User-Agent: curl/7.52.1
Accept: */*
Proxy-Connection: Keep-Alive
Content-Length: 32442
Expect: 100-continue
Content-Type: multipart/form-data; boundary=------------------------082c57537f40911f
</pre>
</blockquote>

<p>Some possible problems are:</p>
<ul>
<li><p>Missing or unknown request method.</p></li>
<li><p>Missing URL.</p></li>
<li><p>Missing HTTP Identifier (HTTP/1.0).</p></li>
<li><p>Request is too large.</p></li>
<li><p>Content-Length missing for POST or PUT requests.</p></li>
<li><p>Illegal character in hostname; underscores are not allowed.</p></li>
<li><p>HTTP/1.1 <q>Expect:</q> feature is being asked from an HTTP/1.0 software.</p></li>
</ul>

<p>Your cache administrator is <a href="mailto:root?subject=CacheErrorInfo%20-%20ERR_INVALID_REQ&amp;body=CacheHost%3A%20%0D%0AErrPage%3A%20ERR_INVALID_REQ%0D%0AErr%3A%20%5Bnone%5D%0D%0ATimeStamp%3A%20Tue,%2012%20Nov%202019%2010%3A46%3A04%20GMT%0D%0A%0D%0AClientIP%3A%2010.24.67.227%0D%0A%0D%0AHTTP%20Request%3A%0D%0APOST%20%2F%20HTTP%2F1.1%0AHost%3A%20ix.io%0D%0AUser-Agent%3A%20curl%2F7.52.1%0D%0AAccept%3A%20*%2F*%0D%0AProxy-Connection%3A%20Keep-Alive%0D%0AContent-Length%3A%2032442%0D%0AExpect%3A%20100-continue%0D%0AContent-Type%3A%20multipart%2Fform-data%3B%20boundary%3D------------------------082c57537f40911f%0D%0A%0D%0A%0D%0A">root</a>.</p>
<br>
</div>

<hr>
<div id="footer">
<p>Generated Tue, 12 Nov 2019 10:46:04 GMT by proxy (squid/3.1.10)</p>
<!-- ERR_INVALID_REQ -->
</div>
</body></html>. Please check the log for anomalies (e.g. swapping
or throttling happenend) and otherwise share this URL.

```

The installation of the tools went fine (I guess thanks to the http_proxy and https_proxy environment variables), but posting the results failed.

Thanks.

ThomasKaiser, 2019-11-12:
> the http_proxy and https_proxy environment variables

Obviously curl is using `http_proxy` as well since the meaningless error message is generated by a Squid proxy server (that seems not to be properly administrated according to 'root' being the cache administrator's address).

gounthar, 2019-11-12:
Unfortunately, I won't be able to bench this machine without a proxy... and I can't change anything to the proxy, of course. 
Is there a report somewhere that I can grab before the scripts removes it?
Thanks.

ThomasKaiser, 2019-11-12:
Simply comment out the [trap statement](https://github.com/ThomasKaiser/sbc-bench/blob/922199e956fed69d75f8759995f1836fed1fe776/sbc-bench.sh#L464) that usually removes the temporary directory (` /tmp/sbc-bench.*`) after execution and pull the result log from there.

closed 2019-11-12 by ThomasKaiser