ThomasKaiser/sbc-bench #10: PINEBOOK pro

chwe17, 2019-12-30:
http://ix.io/25Ut

with stock os

ThomasKaiser, 2020-01-01:
Well, almost same numbers as 2nd RockPi 4B in the [results list](https://github.com/ThomasKaiser/sbc-bench/blob/master/Results.md) (also using the 'strange' combination of combining a 64-bit kernel with a 32-bit user land). Only real differences are broken thermal readouts in your build (18.8°C) and very low AES-128 (16 byte) score.

closed 2020-01-01 by ThomasKaiser

chwe17, 2020-01-02:
I noticed the thermals are broken but didn't had time to look into it. I guess 32bit userspace is mostly due to better supported hardware accelerated decoding. @teacupx mentioned something like that for browsersupport. Overall the pinebook pro looks well made. I'll send you an update once Armbian runs properly on it.

teacupx, 2020-01-02:
Yes, all modern Chromebooks have 64 bit kernel and 32 bit userland. And, since the only way of getting the widevine library to play DRM protected content is extracting it from a Chromebook image, my guess is they decided to go 32 bit, so their users could play video from video streaming sites.

ThomasKaiser, 2020-01-03:
> I'll send you an update once Armbian runs properly on it

Why? This thing runs with a RK3399 and this is all that matters, right? If one SoC appears multiple times in the results list this is only to show how benchmarks depend on software. So what would be the benefit of listing yet another RK3399 result unless there's something seriously broken in the PineBooks hardware design that results in lower scores than usual?

And why should playground projects like Armbian matter?

chwe17, 2020-01-03:
> And why should playground projects like Armbian matter?
Up to everyone to decide, somehoe there's still an user-base which seems to be interested in the project. 

> Why? This thing runs with a RK3399 and this is all that matters, right? If one SoC appears multiple times in the results list this is only to show how benchmarks depend on software.

and ram and thermal design maybe. And if software is kept as close as possible (thereof armbian on all, it makes it more comparable (obviously an armbian from ages ago compared to a recent one will behave different... but at least you have the chance to compare them when rebuilding and retesting.

> listing yet another RK3399 result unless there's something seriously broken in the PineBooks hardware design that results in lower scores than usual?

in case everything runs smooth to show that nothing is broken? It's not only that one must show only broken stuff. Sometimes its just nice to see that a boardvendor did right and delivered a well crafted product. I can't be 100% sure yet, but it seems the pinebook pro might be one of those.