ThomasKaiser/sbc-bench #12: rock64 and vcgencmd

clach04, 2020-02-09:
I knocked up a (poor) vcgencmd https://github.com/clach04/rock64_vcgencmd implementation for rock64 with very limited functionality and it is installed in the same location raspberry pi software expects it to be. Because it is incomplete and there are various checks for its presence, runs on rock64 end up with errors, see http://ix.io/2b8U.

Not sure what the best way to deal with this but I noticed the same check is used in many places rather  than one check and then setting a value to check. I noticed `$CPUs` is set, here https://github.com/ThomasKaiser/sbc-bench/blob/master/sbc-bench.sh#L123 - can that be used through out the script?

I ended up doing a quick hack:

    diff sbc-bench.sh hack_sbc-bench.sh
    121c121,122
    <       if [ -f /usr/bin/vcgencmd ]; then
    ---
    >       #if [ -f /usr/bin/vcgencmd ]; then
    >       if [ -f /usr/bin/raspi-config ]; then
    482c483,485
    <       if [ -f /usr/bin/vcgencmd ]; then
    ---
    >       # FIXME raspberrypi check here instead
    >       #if [ -f /usr/bin/vcgencmd ]; then
    >       if [ -f /usr/bin/raspi-config ]; then
    560c563,565
    <                       if [ -f /usr/bin/vcgencmd ]; then
    ---
    >                       # FIXME raspberrypi check here instead
    >                       #if [ -f /usr/bin/vcgencmd ]; then
    >                       if [ -f /usr/bin/raspi-config ]; then
    768c773,775
    <       if [ -f /usr/bin/vcgencmd ]; then
    ---
    >       # FIXME pi check here instead
    >       #if [ -f /usr/bin/vcgencmd ]; then
    >       if [ -f /usr/bin/raspi-config ]; then

but I do not recommend this approach - this was a quick-and-dirty change to make it work :-)

ThomasKaiser, 2020-02-10:
Well, I'm not planning to deal with this situation since 'faking' any SBC to be a RPi is a bit weird IMO.

clach04, 2020-02-12:
> Well, I'm not planning to deal with this situation since 'faking' any SBC to be a RPi is a bit weird IMO.

I wouldn't disagree, faking an RPi is weird (its also convenient for running some pi software that's not been ported to support other backends) :)

If I sent a PR that had a single file check and then used a global would you be open to that? E.g. something along the lines of:

    if [ -z "${USE_VCGENCMD}" -a -f /usr/bin/vcgencmd ]; then
        USE_VCGENCMD=true
    fi

    #....

    if [ ${USE_VCGENCMD} = true ] ; then
        echo use vcgencmd ....
    fi

    #...
    if [ ${USE_VCGENCMD} = true ] ; then
    #....

that would retain the existing behavior but would only check for vcgencmd  once. It also also allow it to be disabled externally, e.g. `env USE_VCGENCMD=false sh sbc-bench -c`

pfeerick, 2020-02-12:
Having the check overriddable would be good, for when the script incorrectly detects a SBC as a rPi. Better yet would be a better way to detect if it's a rPi than the presence of a file only guaranteed to be supplied with Raspbian. Would it not be better to use a signature such as that exposed at `/proc/device-tree/model` ? Or does that not always get populated by the various rPi images?

clach04, 2020-02-13:
@pfeerick I like the idea of other fingerprinting options (I'm not sure what would be good ones) but consolidating into a single check would make that easier (probably worth a new issue). Sounds like this might be useful for you as well as me so I'll knock something up now.

clach04, 2020-02-13:
PR #13  opened to get discussion going.

I don't have access to any of my pi's nor rock64 boards to hand so this is untested (the fragment above was tested so I'm relatively confident it works).

Discussion points for review:
 * name of variable (e.g. could use something like `DISABLE_VCGENCMD` instead and flip the logic
 * is there a more appropriate Raspberry Pi check? (Could be handled in a separate issue.)

ThomasKaiser, 2020-02-13:
I changed the formatting, position and comments for the functionality and merged the updated PR. 

And I want to point out that I have no interest in getting further attempts in this direction.

`sbc-bench` is not a typical 'fire and forget' benchmark generating silly graphs and numbers. It's a tool meant to generate insights. 

The Raspberries are special in a way that Linux there is a 2nd or 3rd class citizen only. The main (RT)OS called ThreadX runs on the main CPU called VideoCore. ARM/Linux is just a guest here. Therefore it's essential when being able to query ThreadX/VC to do this since the information we can gather from Linux/kernel can't be trusted (like cpufreq for example).

With that in mind if anyone for whatever reasons is keen on using `sbc-bench` on a platform that behaves weird (faked VideoCore interface command lying somewhere -- that's the only real purpose of `vcgencmd`) it's up to him to fix the environment first. By either moving `vcgencmd` out of its location prior to calling `sbc-bench` or now setting some variable.

closed 2020-02-13 by ThomasKaiser

clach04, 2020-02-14:
Thanks @ThomasKaiser looks great, having an option to disable (whilst keeping the fake vcgencmd in place) is what I needed 👍