ThomasKaiser/sbc-bench #17: load > 1 for 7-zip single core benchmark

adatum, 2020-06-29:
Checking the section "System health while running 7-zip single core benchmark" in any result (for example: [here](http://ix.io/1iLy)) shows the load is greater than 1, up to the number of cores in the system, during the single core benchmark. The %cpu correctly matches 100%/[# of cores]. 

It appears the benchmark is being run on a single core, but somehow the benchmark is multithreaded and loading more than one core.

Is this by design? Could it be compromising the results somewhat?

ThomasKaiser, 2020-06-29:
> Is this by design? Could it be compromising the results somewhat?

It's by design and can not compromise the results since those single core results are not even collected. This is just to give users a rough estimate how much faster with 'server loads in general' a big core might be compared to a little one on boards with big.LITTLE SoCs like Exynos 5422 or RK3399.

adatum, 2020-06-29:
The results are shown in the link, and it looks strange to see load > 1 for a single core benchmark, don't you think?

Why is there that substantially increased load at all?

I agree that it is interesting to compare single (little/big) core and multicore results.

ThomasKaiser, 2020-06-29:
> The results are shown in the link, and it looks strange to see load > 1 for a single core benchmark, don't you think?

No, I don't think so. But I always wonder why Linux users seem to be so affected by average load, some number [majority of users not understanding](http://www.brendangregg.com/blog/2017-08-08/linux-load-averages.html)).

The benchmark runs multithreaded (since I have no idea how to tell the 7-zip binary to use only a single core) but it's limited to one single CPU core. So there are `n` threads running simultaneously on one single core and this is perfectly fine to get an idea how much faster with 'server loads in general' a big core might be compared to a little one on boards with big.LITTLE SoCs like Exynos 5422 or RK3399.

closed 2020-06-29 by ThomasKaiser