ThomasKaiser/sbc-bench #21: Comparison dashboard

mrbluecoat, 2021-08-13:
Is there a centralized public dashboard that aggregates user results and displays them in comparison charts, like Phoronix? Your ReadMe examples are great and the raw data posts in these issues are great and the pics folder is nice but the ideal would be a public leaderboard so testers don't have to reinvent the wheel by buying and testing hardware others have tested. Thoughts?

Great project, btw

ThomasKaiser, 2021-08-25:
> Is there a centralized public dashboard that aggregates user results and displays them in comparison charts, like Phoronix? 

Nope. 

Please keep in mind what this repo's README states: 'it tries to produce insights instead of fancy graphs'. And what's written as 3 paragraph in the [results collection](https://github.com/ThomasKaiser/sbc-bench/blob/master/Results.md): 'So do **not** rely on collected numbers unless you carefully read through all the explanations and insights below and be prepared to conduct your own benchmarks if you really want to choose appropriate hardware for **your** use case.'

Why are there several different RockPro64 numbers in the results collection? They're there to help with learning that you never benchmark hardware but also software+settings as well and that you need to be aware of **your** use case to make any good use of collected numbers.

mrbluecoat, 2021-08-25:
I see, thanks

closed 2021-08-25 by mrbluecoat