ThomasKaiser/sbc-bench #26: Phytium D2000 Benchmark

lanefu, 2021-11-25:
|  |  MHz | 5.13 | Impish arm64 | 16720 | 448270 | 828340 | 2962980 | 8158030 | - | [http://ix.io/3FTK](http://ix.io/3FTK) |

Ran by PtitSeb yesterday

ThomasKaiser, 2021-11-25:
A bit sad that this ran with Impish and he didn't manage to build `mhz`. The crypto scores look a bit weird (but since Impish the only other reference collected with `OpenSSL 1.1.1l` and ARMv8 Crypto Extensions is the A73 @ 2.4 GHz on ODROID N2+: http://ix.io/3DtN

            type             16 bytes     64 bytes    256 bytes   1024 bytes   8192 bytes  16384 bytes
    FTC663: aes-256-cbc     386155.67k   644563.61k   774073.94k   808907.43k   826922.33k   828205.74k
    A73:    aes-256-cbc     378114.54k   847673.83k  1187437.06k  1313454.42k  1363293.53k  1366933.50k

closed 2021-11-25 by ThomasKaiser

lanefu, 2021-11-25:
> A bit sad that this ran with Impish and he didn't manage to build `mhz`. The crypto scores look a bit weird (but since Impish the only other references collected with `OpenSSL 1.1.1l` and ARMv8 Crypto Extensions are the A53 @ 2.0 GHz and A73 @ 2.4 GHz on ODROID N2+: http://ix.io/3DtN
> 
> ```
>         type             16 bytes     64 bytes    256 bytes   1024 bytes   8192 bytes  16384 bytes
> FTC663: aes-256-cbc     386155.67k   644563.61k   774073.94k   808907.43k   826922.33k   828205.74k
>    A53: aes-256-cbc     162742.46k   427874.47k   717046.87k   877222.91k   938270.72k   942221.99k
>    A73: aes-256-cbc     378114.54k   847673.83k  1187437.06k  1313454.42k  1363293.53k  1366933.50k
> ```

Yeah some of the #'s seemed weird  He wasn't too familiar with the tool.  Either.   I pre-ordered one.   I'll provide more tests if things work out.

Also FYI if you didn't catch my comment on cnx. It can support up to 32 gigs of RAM so pretty interesting device.

ThomasKaiser, 2021-11-25:
Would be interesting if a dual-channel RAM config results in memory performance getting somewhat decent.

Good luck with this thing. For me it has two huge disadvantages each of them being a noisy fan.

ThomasKaiser, 2021-11-27:
@lanefu are you able to obtain `grep ^CPU /proc/cpuinfo` from the device? I am about to add ARM core details to sbc-bench v0.8.5.

lanefu, 2021-11-27:
K I asked them to run it... will share after they respond.

just an FYI:

I had some back and forth with some honeycomb people about the SBCbench benchmarks.  (I'm relaying info here, no horse in this race... i asked them to open an issue, but wasn't interest in a deeper conversation)

1. Concern that 7zip single core test needed different argument.   was running same number of threads as CPU count, but just on 1 core.    Their point was the `-mnt 1` argument to 7zip should be `-mnt=1`
2. Concern that honeycomb test was at 1.9ghz.  They provided a fresh run, but.... OS info was...fedora?  http://ix.io/3GcC

lanefu, 2021-11-27:
this is from *FT2000/4* which should be the 4 core predecessor to the D2000.. they're gonna give me D2000 info later today 

```
CPU implementer : 0x70
CPU architecture: 8
CPU variant     : 0x1
CPU part        : 0x663
CPU revision    : 3
CPU implementer : 0x70
CPU architecture: 8
CPU variant     : 0x1
CPU part        : 0x663
CPU revision    : 3
CPU implementer : 0x70
CPU architecture: 8
CPU variant     : 0x1
CPU part        : 0x663
CPU revision    : 3
CPU implementer : 0x70
CPU architecture: 8
CPU variant     : 0x1
CPU part        : 0x663
CPU revision    : 3
```

ThomasKaiser, 2021-11-27:
The FT2000 core is already in my list:

    41:Arm
    41/810:ARM810
    41/920:ARM920
    41/922:ARM922
    41/926:ARM926
    41/940:ARM940
    41/946:ARM946
    41/966:ARM966
    41/a20:ARM1020
    41/a22:ARM1022
    41/a26:ARM1026
    41/b02:ARM11 MPCore
    41/b36:ARM1136
    41/b56:ARM1156
    41/b76:ARM1176
    41/c05:Cortex-A5
    41/c07:Cortex-A7
    41/c08:Cortex-A8
    41/c09:Cortex-A9
    41/c0d:Cortex-A17
    41/c0f:Cortex-A15
    41/c0e:Cortex-A17
    41/c14:Cortex-R4
    41/c15:Cortex-R5
    41/c17:Cortex-R7
    41/c18:Cortex-R8
    41/c20:Cortex-M0
    41/c21:Cortex-M1
    41/c23:Cortex-M3
    41/c24:Cortex-M4
    41/c27:Cortex-M7
    41/c60:Cortex-M0+
    41/d01:Cortex-A32
    41/d03:Cortex-A53
    41/d04:Cortex-A35
    41/d05:Cortex-A55
    41/d07:Cortex-A57
    41/d08:Cortex-A72
    41/d09:Cortex-A73
    41/d0a:Cortex-A75
    41/d0b:Cortex-A76
    41/d0c:Neoverse-N1
    41/d0d:Cortex-A77
    41/d0e:Cortex-A76AE
    41/d13:Cortex-R52
    41/d20:Cortex-M23
    41/d21:Cortex-M33
    41/d22:Cortex-M55
    41/d41:Cortex-A78
    41/d42:Cortex-A78AE
    41/d4a:Neoverse-E1
    41/d4b:Cortex-A78C
    42:Broadcom
    42/00f:Broadcom Brahma B15
    42/100:Broadcom Brahma B53
    42/516:Broadcom ThunderX2
    43:Cavium
    43/0a0:Cavium ThunderX
    43/0a1:Cavium ThunderX 88XX
    43/0a2:Cavium ThunderX 81XX
    43/0a3:Cavium ThunderX 83XX
    43/0af:Cavium ThunderX2 99xx
    44:DEC
    44/a10:DEC SA110
    44/a11:DEC SA1100
    46:Fujitsu
    46/001:A64FX
    48:HiSilicon
    48/d01:Kunpeng-920
    49:Infineon
    4d:Motorola/Freescale
    4e:NVidia
    4e/000:NVidia Denver
    4e/003:NVidia Denver 2
    4e/004:NVidia Carmel
    50:APM
    50/000:APM X-Gene
    51:Qualcomm
    51/00f:Qualcomm Scorpion
    51/02d:Qualcomm Scorpion
    51/04d:Qualcomm Krait
    51/06f:Qualcomm Krait
    51/201:Qualcomm Kryo
    51/205:Qualcomm Kryo
    51/211:Qualcomm Kryo
    51/800:Qualcomm Falkor V1/Kryo
    51/801:Qualcomm Kryo V2
    51/c00:Qualcomm Falkor
    51/c01:Qualcomm Saphira
    53:Samsung
    53/001:Samsung Exynos-m1
    56:Marvell
    56/131:Marvell Feroceon 88FR131
    56/581:Marvell PJ4/PJ4b
    56/584:Marvell PJ4B-MP
    61:Apple
    61/022:Apple Icestorm
    61/023:Apple Firestorm
    66:Faraday
    66/526:Faraday FA526
    66/626:Faraday FA626
    69:Intel
    69/200:Intel i80200
    69/210:Intel PXA250A
    69/212:Intel PXA210A
    69/242:Intel i80321-400
    69/243:Intel i80321-600
    69/290:Intel PXA250B/PXA26x
    69/292:Intel PXA210B
    69/2c2:Intel i80321-400-B0
    69/2c3:Intel i80321-600-B0
    69/2d0:Intel PXA250C/PXA255/PXA26x
    69/2d2:Intel PXA210C
    69/411:Intel PXA27x
    69/41c:Intel IPX425-533
    69/41d:Intel IPX425-400
    69/41f:Intel IPX425-266
    69/682:Intel PXA32x
    69/683:Intel PXA930/PXA935
    69/688:Intel PXA30x
    69/689:Intel PXA31x
    69/b11:Intel SA1110
    69/c12:Intel IPX1200
    70:Phytium
    70/662:Phytium FT2000PLUS
    70/663:Phytium S2500
    70/???:Phytium D2000
    c0:Ampere

Now looking into the 7-zip issue...

ThomasKaiser, 2021-11-27:
> Concern that 7zip single core test needed different argument. was running same number of threads as CPU count, but just on 1 core. Their point was the -mnt 1 argument to 7zip should be -mnt=1

They are absolutely right. I tried to fix that over 3 years ago with v0.5.2 but failed since I used `mmt 1` instead of `mmt1` or `mmt=1`. The more cores the CPU is equipped with the more single-threaded results were affected. And not just lower scores but execution time magnitudes longer.

Now in the process of measuring 'wrong' vs 'right' to decide which result list entries need to be thrown away (though I know of noone except myself who looks at those single core scores regularly).

tparys, 2021-11-28:
> The FT2000 core is already in my list (started with `arm-chip-id.db` from @tparys and then adding other stuff I found):

As much as I'd like to take credit for [that database](https://github.com/util-linux/util-linux/blob/master/sys-utils/lscpu-arm.c), it's really part of the util-linux package. If you have new additions, they should probably go there.

lanefu, 2021-11-28:
Okay ptitSeb said D2000 output was identical to F2000, just 8 cores instead
of 4

ThomasKaiser, 2021-11-28:
> ptitSeb said D2000 output was identical to F2000, just 8 cores instead

Which is reasonable and also an indication that `util-linux` people confuse [SoCs with cores](https://en.wikipedia.org/wiki/FeiTeng_(processor)#Future_processors) :)

ThomasKaiser, 2021-11-28:
@tparys I found relying on `regs/identification/midr_el1` not that reliable (missing on some systems where I tested in the meantime). I came up with reading `/proc/cpuinfo` into an array like this:

    ARMTypes=($(awk -F"0x" '/^CPU implementer|^CPU part/ {print $2}' /proc/cpuinfo))

And then like [that](https://github.com/ThomasKaiser/sbc-bench/blob/ffa66030f5cc2faa0fdabf5621505312f5ee4430/sbc-bench.sh#L265-L267)

ThomasKaiser, 2021-11-28:
@lanefu With cda7a50885bab18f1bea6e13733a6012bf84b3ab I removed entries from systems with more than 8 cores from results list. Also replaced few with 6/8 cores.

In case you're still in touch with the Solidrun guys it would be great if they can submit a new result made with v0.8.6 (same for the Phytium of course). And in case you've still access to the AWS instance with 32 N1 cores or this X-Gene thing, then fresh numbers are highly appreciated.

lanefu, 2021-11-29:
New D2000 benchmarks.   Solidrun said he'd try to get to it this week

In case this device is not already represented in official sbc-bench
results list then please
consider submitting it at https://github.com/ThomasKaiser/sbc-bench/issues
with this line:
| | MHz | 5.13 | Impish arm64 | 16680 | 448610 | 828450 | 2920 | 7310 | - |
[http://ix.io/3GrW](http://ix.io/3GrW) |

commit 56c6ce70 (2021-11-25): Phytium D2000 / FTC663 numbers
Closes #26