ThomasKaiser/sbc-bench #31: heztner.com CPX11 (EPYC 2, 2Core 2GB RAM) #2

tonny-sofijan, 2021-12-31:
| AMD EPYC Processor | no cpufreq support | 5.10 | Bullseye x86_64/amd64 | 8420 | 706250 | 873460 | 23630 | 34120 | - | [http://ix.io/3K8L](http://ix.io/3K8L) |

All is well in this round of bench.
Feature request: please output the result as a file too (maybe sbc-bench.log or something?). My remote connection (SSH) disconnect every n-minutes. So when left unattended, I cannot get the result from my VPS. Thanks!

closed 2021-12-31 by ThomasKaiser

ThomasKaiser, 2021-12-31:
`/var/log/sbc-bench.log` now implemented. But even without that 'fix' that would've been a good use for `nohup` since redirecting output to a file and also preventing programs being killed once remote sessions got closed.

Wrt the latter (remote connection (SSH) disconnect every n-minutes). Have you checked your local SSH settings? E.g.

    tk@mac-tk ~ % grep ServerAlive /etc/ssh/ssh_config
    ServerAliveCountMax 900
    ServerAliveInterval 15

tonny-sofijan, 2022-01-03:
Thanks!
As for the `ServerAlive` , if I remember correctly, I've using it in my DO server, but didn't work. I've set it up too at hetzner not too long ago, but it still disconnect. I think the problem is not with the server, but with my ISP. When working from other places (other ISP) from my laptop, no disconnect case.

commit 5419bf6a (2021-12-31): Write continuous /var/log/sbc-bench.log, minor cosmetical fixes.
Fix broken check for x86 to avoid suggesting uploads of x86 results to official list.
Closes #30, closes #31