ThomasKaiser/sbc-bench #33: CPU temperature during system health not detected on (some) Intel and AMD mini PCs

linuxium, 2022-01-04:
After realising the 'missed opportunity' (see my latest comment on 'OS and Memory Impact on Mini PC Gaming Performance') I found that the CPU temperature during system health was not reflected correctly in the latest version of ‘sbc-bench’.

As a temporary work-around I ran the following script:

if $(lsmod | grep -qsw "^k10temp"); then # AMD
        HWMON=$(grep --include=name -rls /sys/devices/pci* -e "^k10temp$" | sed 's?.*hwmon/??' | sed 's?/name??')
elif $(lsmod | grep -qsw "^coretemp"); then # Intel
        HWMON=$(ls /sys/devices/platform/coretemp.0/hwmon)
else
        HWMON=""
fi
if [ -f "/sys/class/hwmon/${HWMON}/temp1_input" ]; then
        echo "sed -i 's?echo 0 >?ln -fs /sys/class/hwmon/${HWMON}/temp1_input ?' sbc-bench.sh"
else
        echo "Cannot determine 'hwmon'"
fi

I'm not suggesting this as 'the fix' but rather using it to highlight where the issue is and one possible way of addressing it.

ThomasKaiser, 2022-01-04:
> the 'missed opportunity'

Thank you for the additional testing (already processing the data). I just didn't respond to you already since asking again for another test (single channel, single rank) seemed a bit too annoying to me. :)

Thank you a lot for the thermal input. I integrated it as follows on x86 ([colorized diff](http://ix.io/3KYF/diff)):

  * try to get the `x86_pkg_temp` `sysfs` node
  * if this fails use your code to search for `$HWMON/temp1_input`
  * if this also fails report at least any sysfs node if existent or otherwise use `0`

Would be great if you can give it a try on an AMD box: http://ix.io/3KYC (`sbc-bench -m` is already sufficient for a test)

On two Intel thingies it looks like this:

    root@gaia:~# sbc-bench.sh -m
    Kernel: x86_64, Userland: amd64
    CPU sysfs topology (clusters, cpufreq members, clockspeeds)
                     cpufreq   min    max
     CPU    cluster  policy   speed  speed   core type
      0        0        0      800    2800   Celeron N5100 @ 1.10GHz
      1        0        1      800    2800   Celeron N5100 @ 1.10GHz
      2        0        2      800    2800   Celeron N5100 @ 1.10GHz
      3        0        3      800    2800   Celeron N5100 @ 1.10GHz
    
    Thermal source: /sys/devices/virtual/thermal/thermal_zone4/temp
    
    Time        CPU    load %cpu %sys %usr %nice %io %irq   Temp
    13:10:26:  884MHz  0.24  13%   3%   9%   0%   0%   0%  42.0°C
    13:10:31:  595MHz  0.22   3%   0%   2%   0%   0%   0%  40.0°C
    ^C
    root@gaia:~# cat /sys/devices/virtual/thermal/thermal_zone4/type 
    x86_pkg_temp
    root@gaia:~# cat /sys/class/hwmon/hwmon2/temp1_input
    40000

    tk@coloc-1:~$ sbc-bench.sh -m
    Kernel: x86_64, Userland: amd64
    CPU sysfs topology (clusters, cpufreq members, clockspeeds)
                     cpufreq   min    max
     CPU    cluster  policy   speed  speed   core type
      0        0        0      800    3000   Xeon Silver 4110 @ 2.10GHz
      1        0        1      800    3000   Xeon Silver 4110 @ 2.10GHz
      2        0        2      800    3000   Xeon Silver 4110 @ 2.10GHz
      3        0        3      800    3000   Xeon Silver 4110 @ 2.10GHz
      4        0        4      800    3000   Xeon Silver 4110 @ 2.10GHz
      5        0        5      800    3000   Xeon Silver 4110 @ 2.10GHz
      6        0        6      800    3000   Xeon Silver 4110 @ 2.10GHz
      7        0        7      800    3000   Xeon Silver 4110 @ 2.10GHz
      8        0        8      800    3000   Xeon Silver 4110 @ 2.10GHz
      9        0        9      800    3000   Xeon Silver 4110 @ 2.10GHz
     10        0       10      800    3000   Xeon Silver 4110 @ 2.10GHz
     11        0       11      800    3000   Xeon Silver 4110 @ 2.10GHz
     12        0       12      800    3000   Xeon Silver 4110 @ 2.10GHz
     13        0       13      800    3000   Xeon Silver 4110 @ 2.10GHz
     14        0       14      800    3000   Xeon Silver 4110 @ 2.10GHz
     15        0       15      800    3000   Xeon Silver 4110 @ 2.10GHz
    
    Thermal source: /sys/devices/virtual/thermal/thermal_zone0/temp
    
    Time        CPU    load %cpu %sys %usr %nice %io %irq   Temp
    13:18:31: 2659MHz  0.82   4%   1%   2%   0%   0%   0%  35.0°C
    13:18:36: 2635MHz  0.75   5%   1%   2%   0%   0%   0%  36.0°C
    ^C
    tk@coloc-1:~$ cat /sys/devices/virtual/thermal/thermal_zone0/type
    x86_pkg_temp
    tk@coloc-1:~$ cat /sys/class/hwmon/hwmon0/temp1_input
    36000

ThomasKaiser, 2022-01-04:
Oh, and it would also be great if you could provide `/sys/class/hwmon/${HWMON}/name` from the AMD box (on Intel it's `coretemp`). Could be used to make the detection more robust.

dan-and, 2022-01-04:
@ThomasKaiser  it's k10temp for AMD cpus since F-Socket to up zen3 

Verified on my AMD Ryzen 5 3600 

See: https://www.kernel.org/doc/html/v5.12/hwmon/k10temp.html

ThomasKaiser, 2022-01-04:
@dan-and thank you! Now I dropped relying on `lsmod` but check `/sys/class/hwmon/hwmon*/name` instead: http://ix.io/3L05 ([diff](http://ix.io/3L06/diff)). Short output from `sbc-bench -m` from an AMD box welcomed. :)

dan-and, 2022-01-04:
sudo ./sbc-bench.sh -m 
Kernel: x86_64, Userland: amd64
CPU sysfs topology (clusters, cpufreq members, clockspeeds)
                 cpufreq   min    max
 CPU    cluster  policy   speed  speed   core type
  0        0        0     2200    3600   AMD Ryzen 5 3600 6-Core
  1        0        1     2200    3600   AMD Ryzen 5 3600 6-Core
  2        0        2     2200    3600   AMD Ryzen 5 3600 6-Core
  3        0        3     2200    3600   AMD Ryzen 5 3600 6-Core
  4        0        4     2200    3600   AMD Ryzen 5 3600 6-Core
  5        0        5     2200    3600   AMD Ryzen 5 3600 6-Core
  6        0        6     2200    3600   AMD Ryzen 5 3600 6-Core
  7        0        7     2200    3600   AMD Ryzen 5 3600 6-Core
  8        0        8     2200    3600   AMD Ryzen 5 3600 6-Core
  9        0        9     2200    3600   AMD Ryzen 5 3600 6-Core
 10        0       10     2200    3600   AMD Ryzen 5 3600 6-Core
 11        0       11     2200    3600   AMD Ryzen 5 3600 6-Core

Time        CPU    load %cpu %sys %usr %nice %io %irq   Temp
16:09:32: 2200MHz  0.99  10%   1%   3%   0%   4%   0%     0°C
16:09:37: 2200MHz  0.91   1%   0%   0%   0%   0%   0%     0°C
16:09:42: 2200MHz  0.84   0%   0%   0%   0%   0%   0%     0°C
16:09:47: 2200MHz  0.77   0%   0%   0%   0%   0%   0%     0°C
16:09:52: 2200MHz  0.71   1%   0%   0%   0%   0%   0%     0°C
16:09:57: 2200MHz  0.65   2%   0%   2%   0%   0%   0%     0°C
16:10:02: 2200MHz  0.60   1%   0%   0%   0%   0%   0%     0°C
16:10:07: 2200MHz  0.55   0%   0%   0%   0%   0%   0%     0°C

$ lscpu | awk -F" " '/^Architecture/ {print $2}'
x86_64

$ cat /sys/devices/virtual/thermal/thermal_zone?/type
iwlwifi_1

$ ls -lF /sys/devices/pci0000:00/0000:00:18.3/hwmon/hwmon1/
total 0
lrwxrwxrwx 1 root root    0 Jan  4 16:08 device -> ../../../0000:00:18.3/
-r--r--r-- 1 root root 4096 Jan  4 16:07 name
drwxr-xr-x 2 root root    0 Jan  4 16:08 power/
lrwxrwxrwx 1 root root    0 Jan  4 16:07 subsystem -> ../../../../../class/hwmon/
-r--r--r-- 1 root root 4096 Jan  4 16:23 temp1_input
-r--r--r-- 1 root root 4096 Jan  4 16:23 temp1_label
-r--r--r-- 1 root root 4096 Jan  4 16:23 temp3_input
-r--r--r-- 1 root root 4096 Jan  4 16:23 temp3_label
-rw-r--r-- 1 root root 4096 Jan  4 16:07 uevent

$ grep '' /sys/devices/pci0000:00/0000:00:18.3/hwmon/hwmon1/* 2>/dev/null 
/sys/devices/pci0000:00/0000:00:18.3/hwmon/hwmon1/name:k10temp
/sys/devices/pci0000:00/0000:00:18.3/hwmon/hwmon1/temp1_input:29500
/sys/devices/pci0000:00/0000:00:18.3/hwmon/hwmon1/temp1_label:Tctl
/sys/devices/pci0000:00/0000:00:18.3/hwmon/hwmon1/temp3_input:29750
/sys/devices/pci0000:00/0000:00:18.3/hwmon/hwmon1/temp3_label:Tccd1

ThomasKaiser, 2022-01-04:
Just out of curiosity... does this work (in reasonable time)?

    find /sys -name temp1_input 2>/dev/null | while read ; do
    	read NodeName <"${REPLY%/*}/name"
    	echo "${REPLY%/*}: ${NodeName} $(cat "${REPLY}")"
    done

Those syfs node names are more self-explanatory / 'telling' than for example `hwmon1` :)

dan-and, 2022-01-04:
$ find /sys -name temp1_input 2>/dev/null | while read ; do
        read NodeName <"${REPLY%/*}/name"
        echo "${REPLY%/*}: ${NodeName} $(cat "${REPLY}")"
done

/sys/devices/pci0000:00/0000:00:18.3/hwmon/hwmon1: k10temp 34625
/sys/devices/pci0000:00/0000:00:01.1/0000:01:00.0/nvme/nvme0/hwmon0: nvme 28850
/sys/devices/virtual/thermal/thermal_zone0/hwmon2: iwlwifi_1 25000

closed 2022-01-04 by ThomasKaiser

linuxium, 2022-01-05:
Seems I'm too late to the party due to doing the testing and not seeing the above updates. Somwwhat now redundant I tested on several Intel mini PCs and a couple of AMD mini PCs. It seems that these newer Intel mini PCs always work using the x86_pkg_temp sensor however I seem to remember before this was not the case.

$ cat sbc-bench.sh.0.9.1.debug
#!/bin/bash

if $(lsmod | grep -qsw "^k10temp"); then # AMD
        HWMON=$(grep --include=name -rls /sys/devices/pci* -e "^k10temp$" | sed 's?.*hwmon/??' | sed 's?/name??')
elif $(lsmod | grep -qsw "^coretemp"); then # Intel
        HWMON=$(ls /sys/devices/platform/coretemp.0/hwmon 2>/dev/null)
fi
if [ -f "/sys/class/hwmon/${HWMON}/temp1_input" ]; then
        echo "DEBUG: Using /sys/class/hwmon/${HWMON}/temp1_input: $(cat /sys/class/hwmon/${HWMON}/temp1_input)"
        if [ -f "/sys/class/hwmon/${HWMON}/temp1_label" ]; then
                echo "DEBUG:       /sys/class/hwmon/${HWMON}/temp1_label: $(cat /sys/class/hwmon/${HWMON}/temp1_label)"
        else
                echo "DEBUG:       /sys/class/hwmon/${HWMON}/temp1_label: file not found"
        fi
        if [ -f "/sys/class/hwmon/${HWMON}/name" ]; then
                echo "DEBUG:       /sys/class/hwmon/${HWMON}/name: $(cat /sys/class/hwmon/${HWMON}/name)"
        else
                echo "DEBUG:       /sys/class/hwmon/${HWMON}/name: file not found"
        fi
fi
$

INTEL
On NGC-5:
linuxium@NGC-5:~/Downloads$ ./sbc-bench.sh.0.9.1.debug
DEBUG: Using /sys/class/hwmon/hwmon3/temp1_input: 40000
DEBUG:       /sys/class/hwmon/hwmon3/temp1_label: Package id 0
DEBUG:       /sys/class/hwmon/hwmon3/name: coretemplinuxium@NGC-5:~/Downloads$ sudo /bin/bash ./sbc-bench.sh.0.9.1 -m
Kernel: x86_64, Userland: amd64
CPU sysfs topology (clusters, cpufreq members, clockspeeds)
                 cpufreq   min    max
 CPU    cluster  policy   speed  speed   core type
  0        0        0      400    4100   Core(TM) i5-8279U @ 2.40GHz
  1        0        1      400    4100   Core(TM) i5-8279U @ 2.40GHz
  2        0        2      400    4100   Core(TM) i5-8279U @ 2.40GHz
  3        0        3      400    4100   Core(TM) i5-8279U @ 2.40GHz
  4        0        4      400    4100   Core(TM) i5-8279U @ 2.40GHz
  5        0        5      400    4100   Core(TM) i5-8279U @ 2.40GHz
  6        0        6      400    4100   Core(TM) i5-8279U @ 2.40GHz
  7        0        7      400    4100   Core(TM) i5-8279U @ 2.40GHz

Thermal source: /sys/devices/virtual/thermal/thermal_zone2/temp

Time        CPU    load %cpu %sys %usr %nice %io %irq   Temp
10:05:12: 4072MHz  0.16   4%   0%   1%   0%   2%   0%  49.0°C
10:05:17: 3906MHz  0.14   0%   0%   0%   0%   0%   0%  35.0°C
^C
linuxium@NGC-5:~/Downloads$ cat /sys/devices/virtual/thermal/thermal_zone1/type
x86_pkg_temp
linuxium@NGC-5:~/Downloads$
On GTi11:
linuxium@GTi11:~/Downloads$ ./sbc-bench.sh.0.9.1.debug
DEBUG: Using /sys/class/hwmon/hwmon3/temp1_input: 41000
DEBUG:       /sys/class/hwmon/hwmon3/temp1_label: Package id 0
DEBUG:       /sys/class/hwmon/hwmon3/name: coretemplinuxium@GTi11:~/Downloads$ sudo /bin/bash ./sbc-bench.sh.0.9.1 -m
Kernel: x86_64, Userland: amd64
CPU sysfs topology (clusters, cpufreq members, clockspeeds)
                 cpufreq   min    max
 CPU    cluster  policy   speed  speed   core type
  0        0        0      400    4200   11th Gen Core(TM) i5-1135G7 @ 2.40GHz
  1        0        1      400    4200   11th Gen Core(TM) i5-1135G7 @ 2.40GHz
  2        0        2      400    4200   11th Gen Core(TM) i5-1135G7 @ 2.40GHz
  3        0        3      400    4200   11th Gen Core(TM) i5-1135G7 @ 2.40GHz
  4        0        4      400    4200   11th Gen Core(TM) i5-1135G7 @ 2.40GHz
  5        0        5      400    4200   11th Gen Core(TM) i5-1135G7 @ 2.40GHz
  6        0        6      400    4200   11th Gen Core(TM) i5-1135G7 @ 2.40GHz
  7        0        7      400    4200   11th Gen Core(TM) i5-1135G7 @ 2.40GHz
[... 116 lines cut ...]
                 cpufreq   min    max
 CPU    cluster  policy   speed  speed   core type
  0        0        0     1400    2300   AMD Ryzen 7 3750H with Radeon Vega Mobile Gfx
  1        0        1     1400    2300   AMD Ryzen 7 3750H with Radeon Vega Mobile Gfx
  2        0        2     1400    2300   AMD Ryzen 7 3750H with Radeon Vega Mobile Gfx
  3        0        3     1400    2300   AMD Ryzen 7 3750H with Radeon Vega Mobile Gfx
  4        0        4     1400    2300   AMD Ryzen 7 3750H with Radeon Vega Mobile Gfx
  5        0        5     1400    2300   AMD Ryzen 7 3750H with Radeon Vega Mobile Gfx
  6        0        6     1400    2300   AMD Ryzen 7 3750H with Radeon Vega Mobile Gfx
  7        0        7     1400    2300   AMD Ryzen 7 3750H with Radeon Vega Mobile Gfx

DEBUG: Using /sys/class/hwmon/hwmon1/temp1_input: 40375
DEBUG:       /sys/class/hwmon/hwmon1/temp1_label: Tctl
DEBUG:       /sys/class/hwmon/hwmon1/name: k10temp
Thermal source: /sys/class/hwmon/hwmon1/temp1_input

Time        CPU    load %cpu %sys %usr %nice %io %irq   Temp
12:53:07: 1400MHz  0.26   3%   0%   2%   0%   0%   0%  40.4°C
12:53:12: 1700MHz  0.24   0%   0%   0%   0%   0%   0%  40.2°C
^C
linuxium@SER3:~/Downloads$ cat /sys/class/hwmon/hwmon1/type
cat: /sys/class/hwmon/hwmon1/type: No such file or directory
linuxium@SER3:~/Downloads$ cat /sys/class/hwmon/hwmon1/name
k10temp
linuxium@SER3:~/Downloads$

linuxium, 2022-01-05:
I've no idea why some of the text above is struck out!

ThomasKaiser, 2022-01-05:
> I've no idea why some of the text above is struck out!

[Markdown](https://docs.github.com/en/github/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#styling-text) at work :) I fixed formatting by using a 4 space `    ` indentation and empty lines before/after.

In case you're booting the NGC-5 one more time can you please provide output of 
`cat /sys/devices/virtual/thermal/thermal_zone2/type`?

linuxium, 2022-01-05:
I'm pretty sure it should read:

linuxium@NGC-5:~/Downloads$ cat /sys/devices/virtual/thermal/thermal_zone2/type
x86_pkg_temp
linuxium@NGC-5:~/Downloads$

I probably introduced a typo as I re-ordered the results into Intel and AMD when I was typing them in which created formatting issues so I started to manually fix them with cut&paste but lost interest as the issue was closed.

commit 1c65e89d (2022-01-04): Try to fix thermal readouts on x86, courtesy @linuxium and @dan-and
Closes #33