ThomasKaiser/sbc-bench #37: Results for SigmaStar SSD202D on SBC2D70 sbc

fifteenhex, 2022-01-21:
| SBC2D70 (SSD202D) | 1700 MHz | 5.16 | Sid armhf | 1950 | 26240 | 33120 | 770 | 3190 | - |  http://ix.io/3MZY |

ThomasKaiser, 2022-01-21:
Hi Daniel, curious: Willy's `mhz` utility shows 1295 MHz while both `/sys/devices/system/cpu/cpufreq/policy0/cpuinfo_max_freq` and `/sys/devices/system/cpu/cpufreq/policy0/scaling_available_frequencies` claim 1700 MHz. Who's right? :)

ThomasKaiser, 2022-01-21:
Can you please provide output of `grep . /sys/devices/system/cpu/cpufreq/policy0/*`?

fifteenhex, 2022-01-21:
```
root@sbc2d70:~# grep . /sys/devices/system/cpu/cpufreq/policy0/*
/sys/devices/system/cpu/cpufreq/policy0/affected_cpus:0 1
/sys/devices/system/cpu/cpufreq/policy0/cpuinfo_cur_freq:599999
/sys/devices/system/cpu/cpufreq/policy0/cpuinfo_max_freq:1700000
/sys/devices/system/cpu/cpufreq/policy0/cpuinfo_min_freq:240000
/sys/devices/system/cpu/cpufreq/policy0/cpuinfo_transition_latency:305000
/sys/devices/system/cpu/cpufreq/policy0/related_cpus:0 1
/sys/devices/system/cpu/cpufreq/policy0/scaling_available_frequencies:240000 400000 600000 800000 1008000 1080000 1188000 1296000 
/sys/devices/system/cpu/cpufreq/policy0/scaling_available_governors:ondemand performance 
/sys/devices/system/cpu/cpufreq/policy0/scaling_boost_frequencies:1350000 1404000 1458000 1512000 1600000 1700000 
/sys/devices/system/cpu/cpufreq/policy0/scaling_cur_freq:599999
/sys/devices/system/cpu/cpufreq/policy0/scaling_driver:cpufreq-dt
/sys/devices/system/cpu/cpufreq/policy0/scaling_governor:ondemand
/sys/devices/system/cpu/cpufreq/policy0/scaling_max_freq:1700000
/sys/devices/system/cpu/cpufreq/policy0/scaling_min_freq:240000
/sys/devices/system/cpu/cpufreq/policy0/scaling_setspeed:<unsupported>
root@sbc2d70:~# 
```

ThomasKaiser, 2022-01-21:
Alright, handling of `scaling_boost_frequencies` is entirely missing. I'm going to prepare a v0.9.2 soon but right now enjoying the good weather for the next hours...

Ah, which `.dts` are you currently using (just to to check OPP nodes there and get a better understanding of what I'm doing)?

fifteenhex, 2022-01-21:
.dts is on the top of this .dtsi with the opp stuff: https://github.com/linux-chenxing/linux/blob/mstar_v5_17_rebase/arch/arm/boot/dts/mstar-infinity2m.dtsi

Currently running with the 1800MHz frequency commented out.

ThomasKaiser, 2022-01-21:
Could you give [v0.9.2](https://github.com/ThomasKaiser/sbc-bench/commit/4fbe5d42fb068cc98cc67d8a43963f4599aa1a37) a try please? 

Would be great if you could also test a quick `sbc-bench -m` on bigboy too with the new version since I hope I got identifying CN8890 right (cluster/NUMA detection still broken though).

fifteenhex, 2022-01-21:
Running now..

fifteenhex, 2022-01-21:
| SBC2D70 (SSD202D) | 1700 MHz | 5.16 | Sid armhf | 1960 | 26240 | 33120 | 770 | 3190 | - | [http://ix.io/3N1U](http://ix.io/3N1U) |

commit d6658d77 (2022-01-22): Few more results and SoC guessing.
Closes #37, closes #38
closed 2022-01-22 by ThomasKaiser