ThomasKaiser/sbc-bench #40: Maybe a way to do the mhz test on gpu?

neofeo, 2022-06-10:
Hi Tkaiser, this project is essential for arm sbc understanding. I was able to prove that my rk3399 was lying on me with the frequencies on manjaro while going further than 2.2 ghz on the big cores. I still dont understand who is thr one cheating here bc on rk3399 that should be driven by the kernel unlike on armlogic platforms and rpi. Maybe it only happens on armbian since I wasnt able to get it further than 2.2 ghz on armbian. I mean, adding another opp to go further.

Since the gpu seems to do the same on rk3399, so, not saying the truth about the freq, would be cool if something like that could be performed on the gpu side of things, since I also noted that the rpi4 gpu also fake his frequencies.

BR, Salvador

I will make a video mentioning this project.

ThomasKaiser, 2022-06-10:
Well, no interest in GPU (on Linux/ARM) at all and therefore no clue anyway. 

Since `sbc-bench` is relying here on @wtarreau's `mhz` utility maybe he has a clue? But I highly doubt it since based on conversations I guess Willy just like me is using his devices headless and can't make much use of GPU capabilities anyway...

wtarreau, 2022-06-10:
I played a little bit with OpenCL a few years ago and thought about that. But that's not practical. First these APIs are horribly complicated, require a complete SDK on the target machine and take ages to build a hello world. Second, the communication timing between the two worlds are very long (especially the boot) and you'd need to run the test for a very long time to amortize that. Third, GPUs do not work like general purpose CPUs and it seems that they're not even always identical, with some having float support and others not for example, and as little as possible is documented about them, so you measure something you have no idea what it's about.

And Thomas is right, I really don't care about such parts that I have no use for and that only serve to heat the SoC :-)

ThomasKaiser, 2022-06-10:
My only 'active/conscious' use for GPU was GPGPU as well: OpenCL in 'Apple packaging' called Grand Central dispatch. We implemented a workflow at a customer doing image comparisons (checking print PDFs for missing elements comparing with screenshots made in the layout software before). A few Mac Minis running the stuff on Intel UHD (or whatever their GPU was called back then) outperformed the same task running on a big fat Solaris server :)

closed 2022-06-10 by ThomasKaiser

neofeo, 2022-06-11:
okay! no problem. https://www.youtube.com/watch?v=wudDU4I-qds