ThomasKaiser/sbc-bench #43: | Allwinner D1 | no cpufreq support | 5.4 | Ubuntu 22.04 LTS riscv64 | 0 | 0 | 0 | 0 | 0 | - |

andypiper, 2022-06-20:
ClockworkPi DevTerm with R01 RISC-V core

andypiper, 2022-06-20:
Note: aware that the tool does not officially support RISC-V, trying things out anyway :-)

andypiper, 2022-06-20:
cpuminer tool does not compile on this board.

ThomasKaiser, 2022-06-20:
> cpuminer tool does not compile on this board

Well, the output doesn't contain anything useful since it seems all benchmarks failed. Can you please provide `dmesg` output from the board after running `sbc-bench`? And also output from `openssl speed -elapsed -evp aes-256-cbc`?

andypiper, 2022-06-20:
This may not be a worthwhile exercise since these boards are seemingly limited right now.

Script output

```
sbc-bench v0.9.8

Installing needed tools. This may take some time. Done.
Checking cpufreq OPP. Done (results will be available in 22-32 minutes).
Executing tinymembench. Done.
Executing RAM latency tester. Done.
Executing OpenSSL benchmark. Done.
Executing 7-zip benchmark. Done.
(/usr/local/src/cpuminer-multi/cpuminer missing or not executable). Done.
Checking cpufreq OPP. Done (1 minutes elapsed).

Memory performance:

7-zip total scores (3 consecutive runs):

OpenSSL results:

Full results uploaded to http://ix.io/40yZ.
```

`dmesg` shows nothing after running:

```
[   34.992464] systemd-journald[100]: Failed to set ACL on /var/log/journal/a4f7484327ea45d0adefc49de6774111/user-1001.journal, ignoring: Operation not supported
[   36.322793] usb1-vbus: disabling
[   43.554057] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[  271.634099] proc: Bad value for 'hidepid'
[  470.267579] proc: Bad value for 'hidepid'
[  593.020273] proc: Bad value for 'hidepid'
[  819.327341] proc: Bad value for 'hidepid'
[  992.774492] bash (22934): drop_caches: 3
[  997.000272] sbc-bench started
```

(then no new output after that)

Here is the `openssl` output:

```
cpi@devterm-R01:~$ openssl speed -elapsed -evp aes-256-cbc
You have chosen to measure elapsed time instead of user CPU time.
Doing AES-256-CBC for 3s on 16 size blocks: 1344179 AES-256-CBC's in 3.00s
Doing AES-256-CBC for 3s on 64 size blocks: 404469 AES-256-CBC's in 3.00s
Doing AES-256-CBC for 3s on 256 size blocks: 105215 AES-256-CBC's in 3.00s
Doing AES-256-CBC for 3s on 1024 size blocks: 26546 AES-256-CBC's in 3.00s
Doing AES-256-CBC for 3s on 8192 size blocks: 3286 AES-256-CBC's in 3.01s
Doing AES-256-CBC for 3s on 16384 size blocks: 1674 AES-256-CBC's in 3.00s
version: 3.0.2
built on: Thu May  5 08:04:52 2022 UTC
options: bn(64,64)
compiler: gcc -fPIC -pthread -Wa,--noexecstack -Wall -g -O2 -ffile-prefix-map=/build/openssl-o9RZ21/openssl-3.0.2=. -fstack-protector-strong -Wformat -Werror=format-security -DOPENSSL_TLS_SECURITY_LEVEL=2 -DOPENSSL_USE_NODELETE -DOPENSSL_PIC -DOPENSSL_BUILDING_OPENSSL -DNDEBUG -Wdate-time -D_FORTIFY_SOURCE=2
CPUINFO: N/A
The 'numbers' are in 1000s of bytes per second processed.
type             16 bytes     64 bytes    256 bytes   1024 bytes   8192 bytes  16384 bytes
AES-256-CBC       7168.95k     8628.67k     8978.35k     9061.03k     8943.16k     9142.27k
```

In order to get the load average down low enough I have to shut off 5-10 services before I can even run the script...

ThomasKaiser, 2022-06-21:
> This may not be a worthwhile exercise since these boards are seemingly limited right now.

Agree but this does not seem to be related to the ISA but something within the OS image. Can you give it another try with these commands please?

    /usr/local/src/ramspeed/ramlat
    echo $PATH
    which openssl
    /bin/bash -x /path/to/sbc-bench.sh

(adjusting `/path/to/` of course :) )

andypiper, 2022-06-21:
```
cpi@devterm-R01:~$ /usr/local/src/ramspeed/ramlat
   size:    1x32    2x32    1x64    2x64   1xPTR   2xPTR   4xPTR   8xPTR
     4k:  240616  166923  249903  198766  499884  471970  177680  120886 
     8k:  237934  188740  231571  200394  493875  470576  243061   93736 
    16k:  239944  182739  248809  184504  489452  454227  241787  117258 
    32k:  172421  179593  241869  198773  415144  462743  234656  117774 
    64k:    5872    5305    6582    5505    6260    6530    5268    4511 
   128k:    6544    5489    6554    5705    5587    6359    6027    4449 
   256k:    6380    4891    6582    5593    6255    6335    6188    4057 
   512k:    6464    5781    5952    5678    5596    6480    4060    4449 
  1024k:    6385    4936    6567    5734    6584    6105    6207    4112 
  2048k:    5884    5150    5660    5165    5141    5823    5515    4076 
  4096k:    5915    4469    5924    5168    5713    5851    4753    4133 
  8192k:    5868    5118    5458    5168    5990    5857    5268    3743 
 16384k:    5683    5008    5719    4381    5793    5656    5075    4021 
cpi@devterm-R01:~$ 
```

```
cpi@devterm-R01:~$ echo $PATH
/home/cpi/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/cpi/.local/bin
```

```
cpi@devterm-R01:~$ which openssl
/usr/bin/openssl
```

<details>
<summary>Full run</summary>

<pre>
cpi@devterm-R01:~$ sudo /bin/bash -x ./sbc-bench.sh
[sudo] password for cpi:
+ Version=0.9.8
+ InstallLocation=/usr/local/src
+ [[ ./sbc-bench.sh = \.\/\s\b\c\-\b\e\n\c\h\.\s\h ]]
+ Main
+ export PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/opt/vc/bin
+ PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/opt/vc/bin
+++ dirname ./sbc-bench.sh
++ cd .
++ pwd -P
+ PathToMe=/home/cpi/sbc-bench.sh
+ unset LC_ALL LC_MESSAGES LANGUAGE LANG
+ test -t 1
++ tput colors
+ ncolors=8
+ test -n 8
+ test 8 -ge 8
++ tput bold
+ BOLD=''
+ NC='\033[0m'
+ LGREEN='\033[1;32m'
+ LRED='\e[0;91m'
++ command -v vcgencmd
+ VCGENCMD=
+ '[' -z '' -a -x '' ']'
+ USE_VCGENCMD=false
+ getopts chmtTpN c
+ CheckRelease
+ '[' -f /etc/os-release ']'
++ awk '-F"' '/^PRETTY_NAME/ {print $2}'
+ OperatingSystem='Ubuntu 22.04 LTS'
+ command -v hostnamectl
++ awk '-F: ' '/Operating System:/ {print $2}'
++ hostnamectl
+ OperatingSystem='Ubuntu 22.04 LTS'
+ grep -q -i Gentoo
[... 605 lines cut ...]
++ head -n1
+ echo -e '\nOpenSSL results\033[0m:\n'

OpenSSL results:

+ grep '^aes-...-cbc' /tmp/sbc-bench.sh.m9kvFA/openssl.log
+ case ${UploadURL} in
+ echo ' http://ix.io/40Af |'
+ echo -e '\nFull results uploaded to http://ix.io/40Af. \c'

Full results uploaded to http://ix.io/40Af. + '[' 0 -le 2 -a 0 -le 5 -a 0 -le 5 -a '!' -f /tmp/sbc-bench.sh.m9kvFA/throttling_info.txt ']'
+ case ${CPUArchitecture} in
+ grep -q '^SoC guess:' /tmp/sbc-bench.sh.m9kvFA/results.log
+ '[' 0 -ne 0 -a X '!=' Xunattended ']'
+ echo

+ '[' -h /var/log/sbc-bench.log ']'
+ '[' -s /var/log/sbc-bench.log ']'
+ echo -e '\n\n\n'
+ cat /tmp/sbc-bench.sh.m9kvFA/results.log
+ BasicSetup
+ rm -rf /tmp/sbc-bench.sh.m9kvFA
+ exit 0
</pre>
</details>

andypiper, 2022-06-21:
Working more fully -> http://ix.io/40BJ

Still seeing some messages about dirname issues in between benchmark executions:

```
Installing needed tools. This may take some time. Done.
Checking cpufreq OPP. Done (results will be available in 48-68 minutes).
Executing tinymembench...dirname: invalid option -- '1'
Try 'dirname --help' for more information.
dirname: missing operand
Try 'dirname --help' for more information.
 Done.
Executing RAM latency tester...dirname: invalid option -- '1'
Try 'dirname --help' for more information.
dirname: missing operand
Try 'dirname --help' for more information.
 Done.
Executing OpenSSL benchmark...dirname: invalid option -- '1'
Try 'dirname --help' for more information.
dirname: missing operand
Try 'dirname --help' for more information.
 Done.
Executing 7-zip benchmark...dirname: invalid option -- '1'
Try 'dirname --help' for more information.
dirname: missing operand
Try 'dirname --help' for more information.
 Done.
Checking cpufreq OPP. Done (22 minutes elapsed).
```

It would be helpful to have a little bit more output when the tool installations / compiles happen, just so it is more obvious what is happening at that point. Just another 2 cents of a thought while I'm here.

closed 2022-06-22 by ThomasKaiser

ThomasKaiser, 2022-06-22:
Thank you for your help!

I won't add these numbers to Results.md since a bit misleading: OpenSSL code is optimized assembler on ARM vs. generic C on RISC-V, I suspect similar optimizations (missing) with 7-zip and so on.

Curious whether there are OS images already working with [mainline kernel](https://github.com/150balbes/build/blob/risc-v/config/sources/riscv64.conf#L26). Seems so: https://forum.armbian.com/topic/21465-armbian-image-and-build-support-for-risc-v/#comment-141582

commit 40b10262 (2022-06-22): Small fixes for Allwinner D1. Closes #43
/sys/devices/system/cpu/cpu0/topology/physical_package_id being -1 caused some troubles