ThomasKaiser/sbc-bench #44: ASUS Tinker Board 2

Necrosys, 2022-06-24:
| ASUS Tinker Board 2/2S | 2016/1512 MHz | 4.4 | Debian GNU/Linux 10 (buster) arm64 | 5920 | 365270 | 1136950 | 3210 | 7500 | 10.66 | http://ix.io/40Nw |

ASUS Tinker Board 2 has been tested with stock heatsink.
OS: [Tinker Board 2 Debian 10 V2.0.4](https://tinker-board.asus.com/download-list.html?product=tinker-board-2#download20220125)

ThomasKaiser, 2022-06-26:
Thank you but Tinkerboard 2 is an uninteresting 2015 design since based on old and boring RK3399 for which plently of results are already listed.

Kernel settings are weird: ASUS guys or whoever they paid for software work use Rockchip's unpatched 4.4 BSP kernel with Android settings like `CONFIG_HZ=1000` while the rest of the (Linux) world switched to `CONFIG_HZ=250`. Memory performance (especially latency) sucks but this is directly related to `CONFIG_HZ` settings. If you know someone at ASUS you could tell them about `CONFIG_ARM_ROCKCHIP_DMC_DEVFREQ` and `CONFIG_HZ=250` to get a better performing system.

But it seems they don't care about software anyway when providing OS images in 2022 with kernel 4.4.194 lacking thousands of (security) fixes. Latest 4.4 LTS version was 4.4.302 from February 3rd this year. Since then 4.4 LTS is EOL and will not receive any more security fixes or as its maintainer said:

> "This kernel branch is now END-OF-LIFE. It will not be getting any more updates from the kernel stable team, and will most likely quickly become insecure and out-of-date. Do not use it anymore unless you really know what you are doing," said Greg Kroah-Hartman. It was a good kernel branch. […] It has powered many millions, maybe a few billion, devices out in the world, but now it’s time to say good-bye.

Providing an OS image with 4.4.194 in 2022 is just insane.

ThomasKaiser, 2022-06-26:
> ASUS Tinker Board 2 has been tested with stock heatsink.

And still throttling occured with each and every of your benchmark runs. Too lazy to compare to other RK3399 numbers to check whether ASUS is using DVFS settings that feed the CPU cores with voltages too high. But since they're famous for not getting thermals right (the RK3288 Tinkerboard was also prone to throttling with their laughable heatsink) it's pointless to look into this direction anyway.

Also basic research about which kind of heatsink sucks less [has been done already](https://github.com/ThomasKaiser/Knowledge/blob/master/articles/Heatsink_Efficiency.md).

closed 2022-06-26 by Necrosys