ThomasKaiser/sbc-bench #47: | Apple MacBook Pro (14-inch, M1 Pro, 2021) | 3036/2064 MHz | 5.18 | Gentoo Base System release 2.8 arm64/aarch64 | 43800 | 620960 | 1064450 | 27110 | 71910 | 48.28 | http://ix.io/443N |

chadmed, 2022-07-11:
Some things to consider:

* The memory controller supports two pstates, but we currently do not activate the higher performance one
* The top pstates for the Firestorm cores are disabled in the devicetree pending changes to PSCI. Firestorm cores _should_ hit 3.2GHz on these machines.

http://ix.io/443N

ThomasKaiser, 2022-07-11:
Thank you!

You've been too fast for me since I just added the new core definitions after you started and as such your benchmark run didn't catch the full set of cluster tests `MODE=extensive` usually runs. But this doesn't matter now. In case you're enabling the higher pstates revisiting this would be great.

Keep up the great work!

closed 2022-07-11 by ThomasKaiser

ThomasKaiser, 2022-07-11:
@chadmed out of curiosity: is there a specific reason to choose `CONFIG_HZ_1000` in your kernel over e.g. `CONFIG_HZ=300`?