ThomasKaiser/sbc-bench #50: sbc-bench.sh with MODE=extensive:  cpuminer build error

slowrunner, 2022-10-22:
On RaspberryPi 4B 2GB, on Ubuntu Server LTS 22.04 (64-bit) spc-bench.sh with MODE=extensive reports: 
```
... (can't build cpuminer) Done.
```
System:
```
Linux ROS2HH 5.15.0-1016-raspi #18-Ubuntu SMP PREEMPT Wed Sep 28 12:15:55 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
```
$ cd /usr/local/src/cpuminer-multi
$ sudo ./build.sh
```
...
crypto/blake2s.c: In function ‘blake2s’:
crypto/blake2s.c:326:9: error: size of array element is not a multiple of its alignment
  326 |         blake2s_state S[1];
      |         ^~~~~~~~~~~~~
make[2]: *** [Makefile:1645: crypto/cpuminer-blake2s.o] Error 1
```

ThomasKaiser, 2022-10-22:
If you're on Debian 11 please try this https://github.com/ThomasKaiser/sbc-bench/issues/43#issuecomment-1161562449 please.

ThomasKaiser, 2022-10-25:
...

slowrunner, 2022-10-25:
No problem - I sent the Rock 4 SE back for a refund and managed to snag another Raspberry Pi4B at PiMoroni, so I'm set for a while.

Thanks for the "support" with benchmarking the Rock SE - sbc-bench was exactly what I needed.

closed 2022-10-25 by slowrunner