ThomasKaiser/sbc-bench #57: ReadSoCTemp function error with Raspberry Pi OS

cnxsoft, 2022-12-30:
I'm running SBC-Bench 0.9.9 in the latest Raspberry Pi OS on the Pi 4, and I get the following error in ReadSoCTemp due to a missing temporary file:

/home/pi/sbc-bench.sh: line 1378: /tmp/sbc-bench.sh.NPKM2L/soctemp: No such file or directory
/home/pi/sbc-bench.sh: line 1379: [: -ge: unary operator expected

ThomasKaiser, 2022-12-30:
Ok, this took a while to reproduce. Partly due to RPi Trading Ltd's moronic attempts to deal with their default user/password mess and then to realize what the problem is.

This only occurs when `sbc-bench` is waiting for CPU utilization and/or average load to settle and it gets stopped by `[ctrl]-[c]`, right? Since then a remaining `sbc-bench -m 5` process running in the background looses the file handle when the parent script has been stopped.

cnxsoft, 2022-12-30:
Ah yes, I think I did press Ctrl+C because the script was waiting with CPU utilization stuck at 5%. I fixed that in my image and restarted the script shortly after.

commit cc0735db (2022-12-30): Some sanity checks wrt background processes
Closes #57
closed 2022-12-30 by ThomasKaiser