ThomasKaiser/sbc-bench #58: Benchmark results for Windows Dev Kit 2023

jglathe, 2023-01-08:
| volterra | ~2980 MHz | 5.15 | arm64 | 33600 | 4789 | 1679480 | 21010 | 41540 | 50.65 | http://ix.io/4kEp |

ThomasKaiser, 2023-01-08:
That's some impressive numbers (this SoC's memory controller rocks). Can you please try to get isolated A78C scores in Linux by executing `taskset -c 4 7zr b -mmt=1` and `taskset -c 4 /usr/local/src/mhz/mhz 3 100000`?

Maybe it ends up with same numbers as for `cpu0` and WSL2 is doing some scheduler optimizations to move single-threaded tasks always to the X1C...

jglathe, 2023-01-08:
I actually doubt that we have this sort of core selectivity in Hyper-V (or WSLVMM). Will try, though. Doing a make with -j8 gets all cores :)

jglathe, 2023-01-08:
jglathe@volterra:~$ taskset -c 4 7zr b -mmt=1

7-Zip (a) [64] 16.02 : Copyright (c) 1999-2016 Igor Pavlov : 2016-05-21
p7zip Version 16.02 (locale=C.UTF-8,Utf16=on,HugeFiles=on,64 bits,8 CPUs LE)

LE
CPU Freq: - - - - - - - - -

RAM size:   15775 MB,  # CPU hardware threads:   8
RAM usage:    435 MB,  # Benchmark threads:      1

                       Compressing  |                  Decompressing
Dict     Speed Usage    R/U Rating  |      Speed Usage    R/U Rating
         KiB/s     %   MIPS   MIPS  |      KiB/s     %   MIPS   MIPS

22:       5310   100   5174   5166  |      56484   100   4839   4823
23:       4563   100   4656   4649  |      55813    99   4857   4831
24:       4362   100   4698   4691  |      54209    99   4806   4759
25:       4134   100   4727   4720  |      52073    98   4723   4635
----------------------------------  | ------------------------------
Avr:             100   4814   4807  |               99   4806   4762
Tot:              99   4810   4784

jglathe@volterra:~$ taskset -c 4 /usr/local/src/mhz/mhz 3 100000
count=645643 us50=10804 us250=54133 diff=43329 cpu_MHz=2980.189
count=645643 us50=10812 us250=54266 diff=43454 cpu_MHz=2971.616
count=645643 us50=10812 us250=54238 diff=43426 cpu_MHz=2973.532

[image]

Compiling Raspberry Pi Linux kernel 6.1.3 took 10 minutes, btw. That's quick. Cross compile on a R9-5900X with all cores available for the VM takes a bout 4:30 minutes.

closed 2023-01-08 by ThomasKaiser

ThomasKaiser, 2023-01-08:
Multi-threaded scores in WSL2 are fine, it's just the limitation that single-threaded the Linux scheduler has no control since Hyper-V or whatever moves single-threaded tasks to a Cortex-X1C (if available).

Thanks for collecting this data! :)

commit 1adc9d8b (2023-01-08): Qualcomm Snapdragon 8cx Gen 3
Closes #58