ThomasKaiser/sbc-bench #60: Microsoft Devkit 2023 results

xlazom00, 2023-03-12:
Microsoft Devkit 2023 results http://ix.io/4qG1

This is native results from linux

ThomasKaiser, 2023-03-13:
Thank you!

I kept the numbers made within WSL2 because interestingly single-threaded performance seems to be a lot better there compared to running Linux 'bare metal' on the box: https://github.com/ThomasKaiser/sbc-bench/blob/3dc3a5dfa7b6aaec187877e9876523af9f750559/results/Sorted-Results.md#7-zip-mips-single-threaded

xlazom00, 2023-03-13:
I will try to run it on my device in WSL2
But it is too much 4789 vs 4283
plz fix that my results are from  Ubuntu 22.10 (kinetic)

ThomasKaiser, 2023-03-13:
> plz fix that my results are from Ubuntu 22.10 (kinetic)

[[x] done](https://github.com/ThomasKaiser/sbc-bench/commit/74eda846673276fa8cf99c8b5c7bc91109d16130).

Maybe it's related to UEFI doing its things in the background (being invisible to anything running with lower privileges like the Linux kernel). Anyway, two things would be interesting: Running the new 'review mode' of `sbc-bench` at least once on your box: `sbc-bench -r` and then repeating the single-threaded 7-zip measurements a few times, e.g.

    for i in 1 2 3 ; do taskset -c 4 7zr b -mmt=1 ; done

xlazom00, 2023-03-13:
I run it in WSL2 
http://ix.io/4qHV

ThomasKaiser, 2023-03-13:
Both times single-threaded 7-zip MIPS north of 4900 so still this strange difference to running bare metal where it's below 4300.

The second run and there both 7-zip multi-threaded MIPS (Total: 31357,27784,23196) as well as cpuminer (39.92 kh/s down to 20.79 kh/s) hint at thermal throttling at work which the Linux inside the WSL2 VM ofc isn't aware of. As such https://github.com/ThomasKaiser/sbc-bench/commit/c134fe77e7697aebde742647e1f8ebe1814e6132 tries to fix the wrong reporting.

Most probably it's an UEFI thing.

Memory latency according to `ramlat` is slightly worse inside the WSL2 VM and only compression scores are affected:

Bare metal:

    RAM size:   31420 MB,  # CPU hardware threads:   8
    RAM usage:    435 MB,  # Benchmark threads:      1
    
                           Compressing  |                  Decompressing
    Dict     Speed Usage    R/U Rating  |      Speed Usage    R/U Rating
             KiB/s     %   MIPS   MIPS  |      KiB/s     %   MIPS   MIPS
    
    22:       4434   100   4324   4314  |      57747   100   4938   4930
    23:       3498   100   3574   3565  |      57082   100   4957   4941
    24:       3175   100   3422   3414  |      56326   100   4953   4945
    25:       2963   100   3392   3384  |      53569   100   4776   4768
    ----------------------------------  | ------------------------------
    Avr:             100   3678   3669  |              100   4906   4896
    Tot:             100   4292   4283

WSL2:

    RAM size:   15775 MB,  # CPU hardware threads:   8
    RAM usage:    435 MB,  # Benchmark threads:      1
    
                           Compressing  |                  Decompressing
    Dict     Speed Usage    R/U Rating  |      Speed Usage    R/U Rating
             KiB/s     %   MIPS   MIPS  |      KiB/s     %   MIPS   MIPS
    
    22:       5587   100   5448   5435  |      57070   100   4889   4873
    23:       4696   100   4793   4785  |      56396    99   4911   4882
    24:       4437   100   4779   4771  |      55932   100   4916   4910
    25:       4268   100   4881   4873  |      53532    98   4862   4765
    ----------------------------------  | ------------------------------
    Avr:             100   4975   4966  |               99   4895   4857
    Tot:             100   4935   4912

And compression speed according to 7-zip author mostly depends on "memory (RAM) latency, Data Cache size/speed and TLB. Also it uses simple 32-bit integer instructions: "shift", "add", "multiply" and other. Out-of-Order execution feature of CPU is also important for that test".

The `ramlat` numbers go only up to 16M but by looking at the `tinymembench` latencies we might get a clue since at 32M and 64M latency bare metal is much worse compared to WSL2:

Bare metal:

       1048576 :    2.1 ns          /     2.0 ns 
       2097152 :   13.4 ns          /    11.1 ns 
       4194304 :   19.9 ns          /    13.9 ns 
       8388608 :   48.3 ns          /    14.4 ns 
      16777216 :   59.7 ns          /    87.0 ns 
      33554432 :  180.8 ns          /   203.3 ns 
      67108864 :  268.7 ns          /   238.5 ns 

WSL2:

       1048576 :    3.3 ns          /     2.2 ns 
       2097152 :   12.3 ns          /    13.2 ns 
       4194304 :   19.9 ns          /    15.8 ns 
       8388608 :   56.5 ns          /    41.9 ns 
      16777216 :   87.0 ns          /   102.6 ns 
      33554432 :  110.6 ns          /   125.4 ns 
      67108864 :  114.8 ns          /   124.8 ns 

BTW: still interested in `sbc-bench -r` output from Linux running bare metal.

xlazom00, 2023-03-13:
sudo /bin/bash ./sbc-bench.sh -r
https://pastebin.com/tK8g0gNW

xlazom00, 2023-03-13:
for i in 1 2 3 ; do taskset -c 4 7zr b -mmt=1 ; done
https://pastebin.com/xwZch2n0

ThomasKaiser, 2023-03-13:
> https://pastebin.com/xwZch2n0

Thank you though we found the culprit earlier: it's the massively increased memory latency with larger access patterns (that's why sbc-bench 0.9.39 will also test through 32M/64M/128M too from then on). Though no idea where this latency increase comes so far other than suspecting something with UEFI (the hardware only partially being under control of the Linux kernel).

BTW: in case time permits you could try to help me fix the OPP table reporting by:

    source sbc-bench.sh ; ParseRawOPPTables | curl -s -F f:1='<-' ix.io

commit 956bc4ac (2023-03-14): Updating results.
Closes #60
Closes #61
closed 2023-03-14 by ThomasKaiser