ThomasKaiser/sbc-bench #61: Results: Quartz64-A under edk2 (UEFI) + Armbian/Debian bookworm generic arm64 UEFI kernel

rpardini, 2023-03-13:
| Pine64 RK3566 Quartz64-A Board | ~1890 | 6.2 | Ubuntu 22.04.2 LTS arm64 | 4980 | 1457 | 884590 | 3240 | 6100 | 6.98 |

http://ix.io/4qJF

ThomasKaiser, 2023-03-13:
Thank you.

Have you tested with the very same board before? Just wondering because this device clocks 5% faster at the 1800 MHz cpufreq OPP. As such it hints even if there's somewhere EDK2 involved there's still RK's boot BLOBs doing PVTM, right?

This is my list of RK3566 results so far (your submission at the end):

    http://ix.io/424k: Cpufreq OPP: 1800    Measured: 1854 (1855.017/1854.442/1854.143)     (+3.0%)
    http://ix.io/4418: Cpufreq OPP: 1800    Measured: 1811 (1816.010/1815.826/1802.363)
    http://ix.io/45X1: Cpufreq OPP: 1800    Measured: 1766 (1766.637/1766.174/1765.768)     (-1.9%)
    http://ix.io/495T: Cpufreq OPP: 1800    Measured: 1700 (1700.916/1700.636/1700.601)     (-5.6%)
    http://ix.io/49Af: Cpufreq OPP: 1800    Measured: 1828 (1829.141/1828.655/1828.614)     (+1.6%)
    http://ix.io/4a00: Cpufreq OPP: 1800    Measured: 1709 (1709.961/1709.784/1709.465)     (-5.1%)
    http://ix.io/4aXR: Cpufreq OPP: 1800    Measured: 1847 (1847.790/1847.088/1846.758)     (+2.6%)
    http://ix.io/4eaW: Cpufreq OPP: 1800    Measured: 1885 (1886.296/1885.866/1885.564)     (+4.7%)
    http://ix.io/4eEr: Cpufreq OPP: 1800    Measured: 1871 (1871.982/1871.940/1871.897)     (+3.9%)
    http://ix.io/4emM: Cpufreq OPP: 1800    Measured: 1833 (1834.012/1833.727/1833.239)     (+1.8%)
    http://ix.io/4fki: Cpufreq OPP: 1800    Measured: 1681 (1682.151/1681.946/1681.877)     (-6.6%)
    http://ix.io/4gms: Cpufreq OPP: 1800    Measured: 1906 (1906.795/1906.619/1906.443)     (+5.9%)
    http://ix.io/4gnG: Cpufreq OPP: 1800    Measured: 1720 (1721.035/1720.928/1720.641)     (-4.4%)
    http://ix.io/4kWK: Cpufreq OPP: 1800    Measured: 1797 (1797.707/1797.199/1797.082)
    http://ix.io/4oAJ: Cpufreq OPP: 1800    Measured: 1762 (1762.963/1762.511/1761.985)     (-2.1%)
    http://ix.io/4qJF: Cpufreq OPP: 1800    Measured: 1893 (1893.470/1893.167/1893.123)     (+5.2%)

With RK's BSP 4.19 kernel at least we got some PVTM information in `dmesg` for example for the [clockspeed 'winner' at 1906 MHz](http://ix.io/4gms):

           cpu cpu0: temp=39444, pvtm=92214 (92240 + -26)
           cpu cpu0: pvtm-volt-sel=1

Unfortunately all this PVTM info got lost with mainline kernel as such we see based on Silicon quality and temperature cpufreq driver claiming RK3566 running at 1800 MHz while in reality it's +/- up to 7%.

closed 2023-03-14 by ThomasKaiser

ThomasKaiser, 2023-03-14:
And since AFAIK you're the father of Armbian's UEFI branch maybe [this here](https://github.com/ThomasKaiser/sbc-bench/issues/60#issuecomment-1466069724) is interesting for you: Armbian/UEFI Ubuntu Kinetic performing worse bare metal on Microsoft's 2023 dev kit compared to a standard Ubuntu VM on exactly same hardware.

Reason already found: with Armbian/UEFI memory latency with larger access patterns is worse compared to the Win11/WSL2 combo. Is there an easy explanation or maybe even a 'fix' for this?

rpardini, 2023-03-15:
> Have you tested with the very same board before? Just wondering because this device clocks 5% faster at the 1800 MHz cpufreq OPP. As such it hints even if there's somewhere EDK2 involved there's still RK's boot BLOBs doing PVTM, right?

Yes, this is using https://github.com/jaredmcneill/quartz64_uefi (master built), which then loads GRUB, which (and this is different than normal) actually loads the mainline quart64a DTB and passes `acpi=off` to kernel. A true frankenstein. Using pure ACPI, there's trouble binding the PHY to the GMAC for the onboard Ethernet, and no way to use anything on the pins (I have some i2c stuff.)

That edk2 build [uses](https://github.com/jaredmcneill/quartz64_uefi/blob/main/build.sh#L75) BL31/DDR from [here](https://github.com/jaredmcneill/quartz64_uefi/tree/main/edk2-rockchip-non-osi/rkbin/bin/rk35) - might be different version and that could explain the 5% diff? 

> And since AFAIK you're the father of Armbian's UEFI branch

Kind-of, fathers actually look after their children, I haven't much; specifically, the kernel config was copied from some 2-year old Ubuntu generic and left to rot... so might be something there.

On the quartz64a, I am now using the [latest stable Kernel PPA debs](https://kernel.ubuntu.com/~kernel-ppa/mainline/v6.2.6/) with great success.

commit 956bc4ac (2023-03-14): Updating results.
Closes #60
Closes #61