ThomasKaiser/sbc-bench #63: Determining Alder Lake N100 performance

ThomasKaiser, 2023-04-07:
(Continuing from https://github.com/Googulator/linux-rk3588-midstream/issues/3#issuecomment-1499862807 )

@n2qcn Thanks for providing the test result over at http://ix.io/4sQP

There are a few things I want to ask though:

  * the AZW MINI S is limited to DDR4 (SO-)DIMMs right? Is there one or two such sockets?
  * Have you looked into dealing with powercap? `powercap-info -p intel-rapl` should tell the limits

Single-threaded the Gracemont core outperforms an A76 in e.g. RK3588 easily due to 3.4 GHz vs. ~2.4 GHz CPU clock (30% more with stuff like 7-ZIP, similar picture with [Geekbench 6](https://browser.geekbench.com/v6/cpu/compare/727918?baseline=798877) but multi-threaded the N100 is bottlenecked by powercapping. The thermal values suggest that [increasing powercap limits](https://www.cnx-software.com/2022/09/08/how-to-check-tdp-pl1-and-pl2-power-limits-in-windows-and-linux/) isn't enough since then throttling might become an issue.

But I would still love to hear whether with a few tweaks performance can be further improved (with this in mind the obvious move to (LP)DDR5 at 4800 MT/s won't be possible though, right?)

n2qcn, 2023-04-07:
the power limit started at 20w, 25w & 78w. Maxing the fan and setting
```
enabled: 1
Zone 0
  name: package-0
  enabled: 1
  max_energy_range_uj: 262143328850
  energy_uj: 4377656395
  Constraint 0
    name: long_term
    power_limit_uw: 78000000
    time_window_us: 27983872
    max_power_uw: 6000000
  Constraint 1
    name: short_term
    power_limit_uw: 78000000
    time_window_us: 2440
    max_power_uw: 0
  Constraint 2
    name: peak_power
    power_limit_uw: 78000000
    time_window_us: 0
    max_power_uw: 0
  Zone 0:0
    name: core
    enabled: 0
    max_energy_range_uj: 262143328850
    energy_uj: 4191830638
    Constraint 0
      name: long_term
      power_limit_uw: 0
      time_window_us: 976
  Zone 0:1
    name: uncore
    enabled: 0
    max_energy_range_uj: 262143328850
    energy_uj: 375670
    Constraint 0
      name: long_term
      power_limit_uw: 0
      time_window_us: 976
```
make it cooler http://ix.io/4sUs but not faster

| Device / details | Clockspeed | Kernel | Distro | 7-zip multi | 7-zip single | AES | memcpy | memset | kH/s |
| ----- | :--------: | :----: | :----: | ----: | ------: | ------: | -----: | -----: | ---: |
| Stock MINI S / N100 | 3400 MHz | 6.1 | Ubuntu 22.04.2 LTS x86_64 | 14010 | 4020 | 1224220 | 9900 | 8900 | - |   
| Loud  MINI S / N100 | 3400 MHz | 6.1 | Ubuntu 22.04.2 LTS x86_64 | 14080 | 4025 | 1232700 | 9980 | 8930 | - |  

Some Alder Lake-N use DDR5, all single channel. Source: https://liliputing.com/intel-introduces-alder-lake-n-series-chips-for-laptops-priced-below-400/

closed 2023-04-14 by ThomasKaiser

n2qcn, 2023-05-09:
N100 with DDR5 http://ix.io/4vxM

| Device / details | Clockspeed | Kernel | Distro | 7-zip multi | 7-zip single | AES | memcpy | memset | kH/s |
| ----- | :--------: | :----: | :----: | ----: | ------: | ------: | -----: | -----: | ---: |
| Stock MINI S / N100 | 3400 MHz | 6.1 | Ubuntu 22.04.2 LTS x86_64 | 14010 | 4020 | 1224220 | 9900 | 8900 | - |   
| Loud  MINI S / N100 | 3400 MHz | 6.1 | Ubuntu 22.04.2 LTS x86_64 | 14080 | 4025 | 1232700 | 9980 | 8930 | - |  
| DDR5 EQ / N100 | 3400 MHz | 6.1 | Ubuntu 23.04 x86_64 | 14150 | 4073 | 1232790 | 11600 | 12270 | 36.24 |