ThomasKaiser/sbc-bench #68: Windows Dev Kit 2023, ubuntu 23.04

jglathe, 2023-06-04:
Hi there,

I'm bringing up my wdk2023, and yep, it boots (although not without errors yet). Anyway, I have multi console login and ssh. It is running Ubuntu 23.04. I need to add lots of firmware to get graphics going, but it should run some tests at least. Here's the result for sbc-bench -G:

Full results uploaded to http://ix.io/4xud

sudo ./sbc-bench.sh -G

Status of performance related governors found below /sys (w/o cpufreq):
3d00000.gpu: simple_ondemand / 270 MHz (userspace powersave performance simple_ondemand / 270 410 500 547 606 640 690)

sbc-bench v0.9.41 taking care of Geekbench

Installing needed tools: apt -f -qq -y install lm-sensors sysstat links, tinymembench, ramlat, mhz, geekbench 6.0.3. Done.
Checking cpufreq OPP. Done.
Executing RAM latency tester. Done.
Executing Geekbench...

Executing on cores 0-3

awk: fatal: error reading input file `-': Device or resource busy
awk: fatal: error reading input file `-': Device or resource busy
/home/jglathe/src/sbc-bench/sbc-bench.sh: line 1576: read: read error: 0: Device or resource busy
/home/jglathe/src/sbc-bench/sbc-bench.sh: line 1577: [: -ge: unary operator expected
Geekbench 6.0.3 Preview : https://www.geekbench.com/

Geekbench 6 requires an active internet connection and automatically uploads
benchmark results to the Geekbench Browser.

Upgrade to Geekbench 6 Pro to enable offline use and unlock other features:

  https://store.primatelabs.com/v6

Enter your Geekbench 6 Pro license using the following command line:

  /usr/local/src/Geekbench-6.0.3-LinuxARMPreview/geekbench6 --unlock <email>
<key>

  Running Gathering system information
System Information
  Operating System              Ubuntu 23.04
  Kernel                        Linux 6.3.0+ aarch64
  Model                         Microsoft Corporation Windows Dev Kit 2023
  Motherboard                   Microsoft Corporation Windows Dev Kit 2023

CPU Information
  Name                          ARM ARMv8
  Topology                      1 Processor, 1 Core, 4 Threads
  Identifier                    ARM implementer 65 architecture 8 variant 0 part 3403 revision 0
  Base Frequency                2.44 GHz

Memory Information
  Size                          30.7 GB

Single-Core
  Running File Compression
  Running Navigation
  Running HTML5 Browser
  Running PDF Renderer
  Running Photo Library
  Running Clang
  Running Text Processing
  Running Asset Compression
  Running Object Detection
  Running Background Blur
  Running Horizon Detection
  Running Object Remover
  Running HDR
  Running Photo Filter
  Running Ray Tracer
[... 266 lines cut ...]
  https://browser.geekbench.com/v6/cpu/1482806/claim?key=6515

Geekbench execution. Done.
Checking cpufreq OPP again. Done (22 minutes elapsed).

First run:

   Single-Core Score     1644
   Multi-Core Score      6807

Second run:

   Single-Core Score     1643
   Multi-Core Score      6813

https://browser.geekbench.com/v6/cpu/compare/1482776?baseline=1482806

Results validation:

  * Measured clockspeed not lower than advertised max CPU clockspeed
  * No swapping
  * Background activity (%system) OK
  * No throttling

Full results uploaded to http://ix.io/4xud

ThomasKaiser, 2023-06-05:
> sudo ./sbc-bench.sh -G

Hi, can you run `./sbc-bench.sh -c` as well (normal mode + `cpuminer`)?

jglathe, 2023-06-05:
jglathe@snapdragix:~/src/sbc-bench$ sudo ./sbc-bench.sh -c
[sudo] password for jglathe:

Status of performance related governors found below /sys (w/o cpufreq):
3d00000.gpu: simple_ondemand / 270 MHz (userspace powersave performance simple_ondemand / 270 410 500 547 606 640 690)

sbc-bench v0.9.41

Installing needed tools, cpuminer. Done.
Checking cpufreq OPP. Done (results will be available in 16-25 minutes).
Executing tinymembench. Done.
Executing RAM latency tester. Done.
Executing OpenSSL benchmark. Done.
Executing 7-zip benchmark. Done.
Executing cpuminer. 5 more minutes to wait. Done.
Checking cpufreq OPP again. Done (16 minutes elapsed).

Results validation:

  * Measured clockspeed not lower than advertised max CPU clockspeed
  * No swapping
  * Background activity (%system) OK
  * No throttling

Memory performance (all 2 CPU clusters measured individually):
memcpy: 14780.6 MB/s (Cortex-A78C)
memset: 38414.1 MB/s (Cortex-A78C)
memcpy: 17503.1 MB/s (Cortex-X1C)
memset: 41781.3 MB/s (Cortex-X1C)

Cpuminer total scores (5 minutes execution): 42.77,42.76,42.75,42.74 kH/s

7-zip total scores (3 consecutive runs): 35229,35417,35454, single-threaded: 4312

OpenSSL results (all 2 CPU clusters measured individually):
type             16 bytes     64 bytes    256 bytes   1024 bytes   8192 bytes  16384 bytes
aes-128-cbc     933707.69k  1542905.77k  1786796.63k  1865112.92k  1885528.06k  1889949.01k (Cortex-A78C)
aes-128-cbc    1311549.58k  2057360.43k  2258969.17k  2307913.05k  2350811.82k  2352360.11k (Cortex-X1C)
aes-192-cbc     880902.59k  1331528.79k  1521115.31k  1574110.21k  1590678.87k  1592776.02k (Cortex-A78C)
aes-192-cbc    1231729.29k  1731241.62k  1904881.58k  1953002.15k  1968472.06k  1972109.31k (Cortex-X1C)
aes-256-cbc     820724.64k  1171806.78k  1311267.93k  1353179.14k  1364363.95k  1365677.40k (Cortex-A78C)
aes-256-cbc    1154209.23k  1535059.80k  1636874.75k  1674321.92k  1685766.14k  1686159.36k (Cortex-X1C)

Full results uploaded to http://ix.io/4xwT

commit 4e758557 (2023-06-06): Some updated numbers.
Closes #68
closed 2023-06-06 by ThomasKaiser