ThomasKaiser/sbc-bench #74: Missing packages for -T option

dominch, 2023-08-08:
Just tried -T option to test some notifications and cpu frequencies on current version and I found out that its not checking requirements. It was done on plain rpi 4b, -T option gives me message about temporary github problem while installing mhz. Inspecting code shows that it just tries to clone that from git and it was easy to find that git is missing on this system. 
I added it via apt-get and then there was warning about missing iostat (line 3049), so I needed to add sysstat package. 
After those everything worked correctly, sbc-bench installed something more but no warnings/errors. 

As far as I remember on other command line switches it adds needed tools, maybe here one another check is needed for those two packages or bins.

ThomasKaiser, 2023-08-08:
Thanks for pointing this out. I chose a quick&dirty fix and just call the `InstallPrerequisits` function that might install a few more packages that are not really needed in this mode. But since this is usually only `dmidecode` and `lm-sensors` with a combined installed size of half a MB I'm fine with this.

closed 2023-08-08 by ThomasKaiser

dominch, 2023-08-09:
That should be ok, probably most users will use more than this mode, so they will trigger this function eventually :) 
Thanks for this and You time :)

commit 1048bb8b (2023-08-08): Install prerequisits also in thermal modes (-t/-T)
Closes #74