ThomasKaiser/sbc-bench #75: Results for Orange Pi 5 w/ legacy kernel

EvilOlaf, 2023-08-10:
Hey Thomas,

not sure if you are still collecting results so feel free to just close here but if you still do here they are:

http://ix.io/4D0a

Not entirely sure anymore if this is already on rkr5.1 or an older tags from Rockchip but it is on 5.10.160 already. Active cooling was in place so low temperatures and no throttling with unlocked to 2.4GHz.

Cheers
Werner

ThomasKaiser, 2023-08-10:
Are these OPP now defaults in Armbian or hand-tuned for your specific RK3588s (which is of very low 'quality' according to the `pvtm-volt-sel=1/2/3` readouts)?

      2304 MHz   1030.0 mV (00ff ffff)
      2352 MHz   1040.0 mV (00ff ffff)
      2400 MHz   1050.0 mV (00ff ffff)

Asking since this is overvolting compared to [Rockchip's original voltage settings](https://github.com/ThomasKaiser/sbc-bench/blob/master/results/opp-tables/rk3588-5.10.66-Radxa_ROCK_5B.txt).

EvilOlaf, 2023-08-10:
Hey,

I did not adjust anything in the dts besides enabling the 2.4GHz OC overlay.

It seems like the overlay was a part of this commit and has been imported from somewhere (?): https://github.com/armbian/linux-rockchip/commit/ea5380bbb7ddf7fdaa2d9717b47e4e2a7957a0d9#diff-8b877f6c903a2a0bd5567399c7877a304593fe259a762b4cd96a59e9599e6f7b

Cheers

ThomasKaiser, 2023-08-10:
> I did not adjust anything in the dts besides enabling the 2.4GHz OC overlay

Ah, ok. Well, this overlay does not exactly 'enable' the 2.4 GHz OPP (this is what the `opp-supported-hw = <0xff 0xffff>;` does, in Rockchip's original sources they highest OPP were denied on low 'quality' PVTM hardware) but does some slight overvolting which will end up with 2.4 GHz on those low 'quality' SoCs but also with clockspeeds north of 2.4 GHz on 'better silicon'. But since the commit is from Jianfeng I'm more than fine with it.

Anyway, now to the question whether to include various RK3588(s) devices – [and there are plenty of them already](https://github.com/ThomasKaiser/sbc-bench/blob/28f2e38cdbb2805bf1797925091bd9f45ebfcf83/sbc-bench.sh#L4934-L4938) – to results list or not. I'm not entirely sure since performance on all RK3588(s) devices is the same since all board makers ship with same LPDDR modules that are bottlenecked by the bootloader to lower speeds or if benchmark scores differ then only caused by PVTM (silicon 'quality') or governor settings (cpufreq and more importantly DMC).

When running with mainline kernel currently performance will suffer since Collabora uses somewhat strange cpufreq/DVFS OPP undervolting the highest cpufreq OPP so the SoCs won't reach their max clockspeeds.

I guess it's reasonable to add your OPi 5 score since this is such a popular device and add a note wrt Rock 5B and OPi 5 being more or less representative for any RK3588(s) device out there.

EvilOlaf, 2023-08-10:
I just got a response and the origin pr is this: https://github.com/armbian/build/pull/4733
Maybe the OPi5 needs slightly higher voltage due to its design? Just pure guess.

Anyway. Cannot call on a decision whether or not to add the results. This is up to you :)

Cheers

closed 2023-08-10 by ThomasKaiser

ThomasKaiser, 2023-08-10:
> Maybe the OPi5 needs slightly higher voltage due to its design?

Nope, Xunlong just shopped low quality RK3588s ([their SoCs are also more recent since they have chip revision 35881000 while everybody else got 35880000](https://github.com/ThomasKaiser/sbc-bench/blob/28f2e38cdbb2805bf1797925091bd9f45ebfcf83/sbc-bench.sh#L4934-L4938)). 

But the RK3588s batch Khadas got from Rockchip for example is similar or even worse. [Various Khadas Edge2 under full load then clock the A76 cores with only ~400 MHz](https://forum.khadas.com/t/suggested-settings-optimisation-for-edge2/16885/2?u=tkaiser) which is a sign of something being fishy with cpufreq/pvtm code. The solution for the Khadas problem would be to remove highest cpufreq OPP since when being limited to 2.35 GHz they can reach and maintain that clockspeed unlike at 2.4 GHz.

commit 92e0cb44 (2023-08-10): Add Orange Pi 5 numbers and RK35xx explanation
Closes #75