ThomasKaiser/sbc-bench #76: Inquiry about ATF on Rockpro64 and freqs

neofeo, 2023-08-21:
Hi TKaiser, I am a bit disappointed with my Rockpro64 on armbian lying to me even with ATF enabled.
Could you explain me why this happened??

[image]

ThomasKaiser, 2023-08-21:
Where's the ix.io link for this?

neofeo, 2023-08-21:
http://ix.io/4DnZ

ThomasKaiser, 2023-08-21:
That's the culprit:

    vdd_cpu_b: 1288 mV (1500 mV max)

vs.

    2400 MHz   1290.0 mV

Voltage feeded to the A72 cores is limited to 1288 mV as such every DVFS OPP that exceeds this won't be activated. 

Asides that the OPP table is weird anyway (the voltage increases should happen in tiny steps at low frequencies and have to get the larger the higher the frequency (see [rk3399-4.4.211-FriendlyElec_NanoPi_M4.txt](https://github.com/ThomasKaiser/sbc-bench/blob/master/results/opp-tables/rk3399-4.4.211-FriendlyElec_NanoPi_M4.txt) for example or any other vendor OPP table there) and this is the wrong place to ask about device-tree basics especially if it's about something that weird as 'overclocking'.

closed 2023-08-21 by ThomasKaiser

neofeo, 2023-08-21:
Thanks for the feedback!

neofeo, 2023-08-21:
I changed my entire opp tables, not they are a lot better, very stable at 2.2/1.8 , didn't noticed that seems thst only the first microvolt valye its being taken in consideration. My low power cores were trying to reach 1.8 with 1.2v and that wasn't  enough, it was required 1.25 or so. Also, who manage that limit on the voltages, bc it mentions maximun values but at the end it limits them way lower, I am not planning to go to much farther than 1.3v, it's  risky, I know, but I will need 1.31 or 1.32 to reach 2.4 ghz.