ThomasKaiser/sbc-bench #77: Results for Raspberry Pi 5 model B using 16k page size

geerlingguy, 2023-09-28:
Firing it up now... just had to wait for the embargo to lift!

```
pi@pi5:~/Downloads $ sudo /bin/bash ./sbc-bench.sh -j
Starting to examine hardware/software for review purposes...

Average load and/or CPU utilization too high (too much background activity). Waiting...

Too busy for benchmarking: 02:35:04 up 0 min,  4 users,  load average: 1.07, 0.31, 0.10,  cpu: 28%
Too busy for benchmarking: 02:35:09 up 0 min,  4 users,  load average: 0.98, 0.30, 0.10,  cpu: 1%
Too busy for benchmarking: 02:35:14 up 0 min,  4 users,  load average: 0.90, 0.30, 0.10,  cpu: 0%
Too busy for benchmarking: 02:35:19 up 0 min,  4 users,  load average: 0.91, 0.31, 0.11,  cpu: 0%
Too busy for benchmarking: 02:35:24 up 0 min,  4 users,  load average: 0.84, 0.30, 0.11,  cpu: 0%
Too busy for benchmarking: 02:35:29 up 1 min,  4 users,  load average: 0.77, 0.30, 0.11,  cpu: 0%
Too busy for benchmarking: 02:35:34 up 1 min,  4 users,  load average: 0.71, 0.29, 0.10,  cpu: 0%

sbc-bench v0.9.47

Installing needed tools: apt -f -qq -y install lm-sensors sysstat lshw mmc-utils smartmontools, tinymembench, ramlat, mhz, cpuminer...
```

I'll report back when there's something to report.

geerlingguy, 2023-09-28:
http://ix.io/4HDw

geerlingguy, 2023-09-28:
```
# Raspberry Pi 5 Model B Rev 1.0

Tested with sbc-bench v0.9.47 on Thu, 28 Sep 2023 02:48:00 -0500. Full info: [http://ix.io/4HDw](http://ix.io/4HDw)

### General information:

    Kernel: aarch64, Userland: arm64
    
    CPU sysfs topology (clusters, cpufreq members, clockspeeds)
                     cpufreq   min    max
     CPU    cluster  policy   speed  speed   core type
      0        0        0     1000    2400   Cortex-A76 / r4p1
      1        0        0     1000    2400   Cortex-A76 / r4p1
      2        0        0     1000    2400   Cortex-A76 / r4p1
      3        0        0     1000    2400   Cortex-A76 / r4p1

8053 KB available RAM

### Governors/policies (performance vs. idle consumption):

Original governor settings:

    cpufreq-policy0: ondemand / 2400 MHz (conservative ondemand userspace powersave performance schedutil / 1000 1100 1200 1300 1400 1500 1600 1700 1800 1900 2000 2100 2200 2300 2400)

Tuned governor settings:

    cpufreq-policy0: performance / 2400 MHz

Status of performance related policies found below /sys:

    /sys/module/pcie_aspm/parameters/policy: default [performance] powersave powersupersave

### Clockspeeds (idle vs. heated up):

Before at 34.2°C:

    cpu0 (Cortex-A76): OPP: 2400, ThreadX: 2400, Measured: 2398 

After at 63.9°C:

    cpu0 (Cortex-A76): OPP: 2400, ThreadX: 2400, Measured: 2398 

### Performance baseline

  * memcpy: 5257.3 MB/s, memchr: 13627.2 MB/s, memset: 11521.6 MB/s
  * 16M latency: 122.1 119.5 125.8 128.5 122.6 123.9 133.9 153.4 
  * 128M latency: 139.6 138.4 139.6 138.4 139.3 138.1 139.3 140.9 
  * 7-zip MIPS (3 consecutive runs): 10816, 10984, 11053 (10950 avg), single-threaded: 3163
  * `aes-256-cbc     590155.97k  1049951.30k  1273231.10k  1338685.78k  1365423.45k  1367943.85k`
  * `aes-256-cbc     590411.13k  1050946.97k  1273904.47k  1340496.55k  1365346.99k  1368036.69k`

### PCIe and storage devices:

  * Vendor Device 0001: Speed 5GT/s, Width x4, driver in use: rp1
  * 119.1GB "SanDisk SR128" UHS SDR104 SDXC card as /dev/mmcblk0: date 10/2019, manfid/oemid: 0x000003/0x5344, hw/fw rev: 0x8/0x0

### Swap configuration:

  * /var/swap on /dev/mmcblk0p2: 100.0M (0K used) on ultra slow SD card storage

### Software versions:

  * Debian GNU/Linux 12 (bookworm)
  * Build scripts: http://archive.raspberrypi.org/debian/ bookworm main
  * Compiler: /usr/bin/gcc (Debian 12.2.0-14) 12.2.0 / aarch64-linux-gnu
  * ThreadX: 9d494316 / 2023/09/13 11:37:06 

### Kernel info:

  * `/proc/cmdline: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1  smsc95xx.macaddr=B8:27:EB:AE:74:3B vc_mem.mem_base=0x3fc00000 vc_mem.mem_size=0x40000000  console=ttyAMA10,115200 console=tty1 root=PARTUUID=22c2ce57-02 rootfstype=ext4 fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles cfg80211.ieee80211_regdom=US`
  * Vulnerability Spec store bypass:    Mitigation; Speculative Store Bypass disabled via prctl
  * Vulnerability Spectre v1:           Mitigation; __user pointer sanitization
  * Vulnerability Spectre v2:           Mitigation; CSV2, BHB
  * Kernel 6.1.47-v8_16k+ / CONFIG_HZ=250

Kernel 6.1.47 is not latest 6.1.55 LTS that was released on 2023-09-23.

See https://endoflife.date/linux for details. Perhaps some kernel bugs have
been fixed in the meantime and maybe vulnerabilities as well.

All known settings adjusted for performance. Device now ready for benchmarking.
Once finished stop with [ctrl]-[c] to get info about throttling, frequency cap
and too high background activity all potentially invalidating benchmark scores.
All changes with storage and PCIe devices as well as suspicious dmesg contents
will be reported too.
```

ThomasKaiser, 2023-09-28:
Thank you. Most interesting bit (at least for me) is the I/O situation: The 40nm RP1 chip is attached via PCIe Gen2 x4 to the SoC. And since german c't magazine measured 3.25W in idle this is just horrible :)

closed 2023-09-28 by ThomasKaiser