ThomasKaiser/sbc-bench #79: Update/Add benchmark data for RockjPro64 and R1505G

Tharadash, 2023-10-03:
As requested in PR #78, here is some fresh benchmark data on the old-and-boring RK3399 in it's Pine64 RockPro64 form as well as a not-so-new-but-interesting AMD Ryzen Embedded R1505G in a HP T640 system.
[4HV0.txt](https://github.com/ThomasKaiser/sbc-bench/files/12788596/4HV0.txt)
[4HVb.txt](https://github.com/ThomasKaiser/sbc-bench/files/12788597/4HVb.txt)
[4HYd.txt](https://github.com/ThomasKaiser/sbc-bench/files/12788598/4HYd.txt)

Thanks!

ThomasKaiser, 2023-10-04:
Just a question... is there a particular reason to not activate the device tree overlays on RockPro 64 to let the cores operate at 2.0/1.5 GHz?

Tharadash, 2023-10-04:
> Just a question... is there a particular reason to not activate the device tree overlays on RockPro 64 to let the cores operate at 2.0/1.5 GHz?

Thanks for asking!
I did not deem it "relevant enough" because armbian does not support that out of the box [since at least last year.](https://forum.armbian.com/topic/20584-armbian-config-it-does-not-work/?do=findComment&comment=139985)

ThomasKaiser, 2023-10-04:
Oh my... Armbian...

In case time permits I would prefer retesting with 2.0/1.5 GHz (only on 6.1 LTS, 6.4 doesn't differ and the results collection is already packed with RK3399 numbers). And IMO there's no point to add 1.8/1.4 GHz results (maybe only to proof my point from the past that with Armbian updating your OS often results in worse performance).

The Ryzen numbers [are already added](https://github.com/ThomasKaiser/sbc-bench/commit/4f1a4c82993bffe9d1380be463e82c966b5ffcb5)

commit 82573ed5 (2023-10-28): Orange Pi 4 scores with kernel 6.1 / Bookworm
Closes #79 since now we have RK3399 scores with kernel 6.1 at 2010/1510 MHz (and all RK3399 boards perform the same except type of DRAM since DDR and LPDDR run with slightly different timings)
closed 2023-10-28 by ThomasKaiser