ThomasKaiser/sbc-bench #86: Unable to compile tinymembench

dominch, 2024-01-09:
I just fired newest "sbc-bench -r" on radxa rock-3b official cli debian 
It failed that cannot install needed lib 
on second try it was more specific: 
```
root@rock-3b:/home/radxa# ./sbc-bench.sh -r                                                                            
Starting to examine hardware/software for review purposes...                                                           
                                                                                                                       
sbc-bench v0.9.60                                                                                                      
                                                                                                                       
Installing needed tools: tinymembench...                                                                                                         
Temporary Github problem. Not able to download tinymembench. Please try again later.                                   
```
I checked sources and it seems to replace one tinymembench with another, 
It successfully cloned github repo but failed to build it and producing this error when github is working correctly 
I removed pipe to /dev/null on build and got this: 
```
root@rock-3b:/usr/local/src/tinymembench# make    
cc -O2  -c util.c  
util.c:24:10: fatal error: string.h: No such file or directory
   24 | #include <string.h>
      |          ^~~~~~~~~~
compilation terminated. 
make: *** [Makefile:23: util.o] Error 1
```
which is easy to solve by ```apt-get install libc6-dev```

So basically on first run it tried to replace one tinymembench with another but failed at build then probably it failed at cloning same thing to same place indicating wrong error about github connectivity. Maybe it should just check for this package before trying to build tinymembench or more complex - for needed functionality. For sure curent error is misleading, github is not a problem.

ThomasKaiser, 2024-01-09:
What is the output of `cat /etc/os-release`?

dominch, 2024-01-09:
Here it is: 
```
root@rock-3b:/home/radxa#  cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

dominch, 2024-01-09:
and: 
```
root@rock-3b:/home/radxa# cat /etc/radxa_image_fingerprint
RBUILD_BUILD_DATE='Thu, 12 Oct 2023 10:55:04 +0000'
RBUILD_REVISION='85da966bd60dbe3e053b0084fe9ec570fcdaeb57'
RBUILD_COMMAND='./rbuild --timestamp=b18 --compress --native-build --shrink rock-3b bullseye cli'
RBUILD_KERNEL='linux-image-5.10.160-18-rk356x'
RBUILD_KERNEL_VERSION='5.10.160-18-425b07550'
RBUILD_UBOOT='u-boot-latest'
RBUILD_UBOOT_VERSION='2023.07.02-6-4257d241'
```