ThomasKaiser/sbc-bench #90: Sipeed Lichee Console 4A

geerlingguy, 2024-03-22:
FYI, I've run sbc-bench on a Lichee Console 4A, Sipeed's RISC-V 'cyberdeck'-style mini notebook running a TH1520 SoC, 4 cores at 1.5 GHz.

Results: https://sprunge.us/fUCnrY

Testing as part of https://github.com/geerlingguy/sbc-reviews/issues/39

ThomasKaiser, 2024-03-22:
Thanks! But is there a reason the SoC only clocking in at 1.5 GHz?

They advertised the TH1520 being capable of 2.5 GHz but all I've seen so far is 2.0 GHz: https://github.com/ThomasKaiser/sbc-bench/blob/master/results/4xYE.txt

ThomasKaiser, 2024-03-22:
It seems they did only limit CPU core clockspeeds but not memory controller:

| Clockspeed | 7-zip multi | 7-zip single | AES | memcpy | memset |
| ----- | ----: | ------: | ------: | -----: | -----: |
| 1500 MHz | 3880 | 1129 | 33260 | 3290 | 11160 |
| 2000 MHz | 5260 | 1592 | 43820 | 4350 | 14760 |

closed 2024-03-22 by ThomasKaiser

geerlingguy, 2024-03-23:
@ThomasKaiser - Indeed, this little thing gets toasty, even at the lower clocks. I think thermals aren't great on the current design, it's definitely a bit of a 'ragtag' developer machine. Decently put together, still (in comparison to some devices I've used at the early revisions), but there's just a heat pipe hanging out in the open space, and fan kinda blowing air at one end of it :D

commit 7e9e5707 (2024-03-22): Add Lichee Console 4A numbers.
Closes #90