ThomasKaiser/sbc-bench #91: Results for Orange Pi 5 Plus (32GB)

martin-zs, 2024-04-13:
https://sprunge.us/Vo25Uv
Only passive cooling which was provided by the Aluminum "Orange Pi 5 Plus Metal Protective Case", the case with the two antennas.

The OS installed on the SD card was "Armbian_24.2.1_Orangepi5-plus_jammy_legacy_5.10.160_xfce-amazingfated_desktop.img.xz" Release date: Feb 13, 2024

ThomasKaiser, 2024-04-14:
Thank you. I only added the review info since performance numbers are the same as with every other RK3588 with LPDDR4X memory :)

With LPDDR5 this might change but not know since [right now there's something wrong with LPDDR5 initialization](https://forum.radxa.com/t/rock-5-in-itx-form-factor/20005/37?u=tkaiser).

closed 2024-04-14 by ThomasKaiser

martin-zs, 2024-04-15:
Thanks,
Just noticed that there were two problems with the last run:
  * Advertised vs. measured max CPU clockspeed: -1.5% before, -2.2% after
  * Too much other background activity: 0% avg, 3% max

Here is new rune after a cold boot without those errors
https://sprunge.us/ct7Oli

ThomasKaiser, 2024-04-15:
> Advertised vs. measured max CPU clockspeed: -1.5% before, -2.2% after

This is not really an 'error' on RK3588 but just to be expected since 'silicon lottery'. The A76 clusters that are officially advertised as 2400 MHz capable on your board stay at either 2.3 and 2.35 GHz:

    Cpufreq OPP: 2304    Measured: 2303 (2303.332/2303.332/2303.074)
    Cpufreq OPP: 2352    Measured: 2337 (2337.624/2337.507/2337.361)

Explained here: https://github.com/ThomasKaiser/Knowledge/blob/master/articles/Quick_Preview_of_ROCK_5B.md#pvtm

commit 87fa4ae9 (2024-04-14): Add review info for Orange Pi 5 Plus.
Closes #91