ThomasKaiser/sbc-bench #95: `intel_pstate` driver and MaxKHz parameter

mlincett, 2024-06-10:
It seems that trying to control the maximum clockspeed through the `MaxKHz` parameter is somewhat ineffective on a platform that uses the `intel_pstate` scaling driver. [An example](https://0x0.st/Xbd2.bin).

Besides, the `/sys/devices/system/cpu/cpufreq/policy?/scaling_available_frequencies` property is not exposed by `intel_pstate`.

Is there a workaround to run the benchmark at a fixed max clockspeed on Intel?

ThomasKaiser, 2024-06-11:
> Is there a workaround to run the benchmark at a fixed max clockspeed on Intel?

Unfortunately [not that I know of](https://www.cnx-software.com/2024/05/01/geekom-xt12-pro-review-ubuntu-24-04-on-an-intel-core-i9-12900h-mini-pc/#comment-617241).

mlincett, 2024-06-11:
I fiddled a bit more.

Running `cpupower frequency-set -u $freqKHz` before `sbc-bench` seems effective in limiting the max clockspeed (see [attachment](https://github.com/user-attachments/files/15783896/Lenovo_M720q_IPC.txt)) when `sbc-bench` is run with an appropriate `$MaxKHz` parameter.

In my case what I tried was `cpupower frequency-set -u 1000000` and `$MaxKHz=800000` for `sbc-bench`, I am not sure which of the two values ended up determining the effective clockspeed (measured at ~990 MHz). If `sbc-bench` is run unconstrained, then the limit set by `cpupower` is overridden in the `CheckCPUCluster()` loop.

ThomasKaiser, 2024-06-11:
Can you please provide output of the following with that specific setup (800 MHz max cpufreq OPP while running with 1000 MHz in reality)?

    grep -r . /sys/devices/system/cpu/*/cpufreq/* ; grep -r . /sys/devices/system/cpu/cpufreq/*

mlincett, 2024-06-11:
> Can you please provide output of the following with that specific setup (800 MHz max cpufreq OPP while running with 1000 MHz in reality)?
> 
> ```
> grep -r . /sys/devices/system/cpu/*/cpufreq/* ; grep -r . /sys/devices/system/cpu/cpufreq/*
> ```

See [attached dump](https://github.com/user-attachments/files/15792885/cpufreq_dump.txt).