ThomasKaiser/sbc-bench #98: Results for Radxa X4 (Preliminary)

geerlingguy, 2024-07-31:
Using stock install of Ubuntu 24.04 Desktop, and running with the official Radxa X4 Heatsink case, here are my results:

[TODO](https://0x0.st/XO8j.bin)

I believe the CPU hits thermal limits as it's hitting 95°C very quickly on any benchmark run, so I may work on a more robust cooling setup, to make sure that's not the problem with some of the lower performance I'm seeing compared to other N100 systems (and not just whatever power profile is set up).

I saw there was an 'X4L' run posted in [this Radxa community forum post](https://forum.radxa.com/t/new-n100-intel-board-x4/21928/48?u=geerlingguy) but I'm not sure if that's an entirely different hardware product, or was a typo for the X4...

More of my benchmarking/testing here: https://github.com/geerlingguy/sbc-reviews/issues/48

geerlingguy, 2024-07-31:
Not much difference (slight improvements across the board), but temps were down about 10° using paste instead of pads, even with a notable gap, about 1mm or maybe 0.6mm between the heatsink metal and the die.

ThomasKaiser, 2024-08-28:
Closed by https://github.com/ThomasKaiser/sbc-bench/commit/76420fff0913e7626224879291295b1d63dc4fcd

closed 2024-08-28 by ThomasKaiser