ThomasKaiser/sbc-bench #99: EIC7700X Results

Fishwaldo, 2024-08-23:
I just got the EVB board from eswin for the new EIC7700x CPU. Results are here:
https://0x0.st/Xyfg.bin

| ESWIN EIC7700 | 1400 MHz | 6.6 | Debian GNU/Linux trixie/sid riscv64 | 5390 | 1485 | 49340 | 4570 | 8700 | - | https://0x0.st/Xyfg.bin |

ThomasKaiser, 2024-08-25:
Thank you! `lm-sensors` reports temperatures of stuff like CPU/GPU but no corresponding `sysfs` seem to be defined.

Are you able to provide output of `cat /sys/class/hwmon/hwmon?/name` in search of a hwmon node representing at least the CPU temperature?

Fishwaldo, 2024-08-25:
Hi,
it has a weird setup (or maybe its just the BSP kernel drivers?) - hwmon6 is the CPU temp, as well as voltages etc:

```total 0
drwxr-xr-x 3 root root    0 Aug 25 23:37 .
drwxr-xr-x 3 root root    0 Aug 25 23:37 ..
lrwxrwxrwx 1 root root    0 Aug 24 01:54 device -> ../../../50b00000.pvt
-r--r--r-- 1 root root 4.0K Aug 25 23:37 in0_input
-r--r--r-- 1 root root 4.0K Aug 25 23:37 in0_label
-r--r--r-- 1 root root 4.0K Aug 25 23:37 in1_input
-r--r--r-- 1 root root 4.0K Aug 25 23:37 in1_label
-r--r--r-- 1 root root 4.0K Aug 25 23:37 in2_input
-r--r--r-- 1 root root 4.0K Aug 25 23:37 in2_label
-r--r--r-- 1 root root 4.0K Aug 25 23:37 in3_input
-r--r--r-- 1 root root 4.0K Aug 25 23:37 in3_label
-r--r--r-- 1 root root 4.0K Aug 25 23:37 label
-r--r--r-- 1 root root 4.0K Aug 25 23:37 name
lrwxrwxrwx 1 root root    0 Aug 25 23:37 of_node -> ../../../../../../firmware/devicetree/base/soc/pvt@0x50b00000
drwxr-xr-x 2 root root    0 Aug 25 23:37 power
lrwxrwxrwx 1 root root    0 Jan  1  1970 subsystem -> ../../../../../../class/hwmon
-r--r--r-- 1 root root 4.0K Aug 25 23:37 temp1_input
-r--r--r-- 1 root root 4.0K Aug 25 23:37 temp1_label
-rw-r--r-- 1 root root 4.0K Aug 25 23:37 temp1_offset
-r--r--r-- 1 root root 4.0K Aug 25 23:37 temp1_type
-rw-r--r-- 1 root root 4.0K Aug 25 23:37 uevent
-rw-r--r-- 1 root root 4.0K Aug 25 23:37 update_interval
root@rockos-eswin:/sys/class/hwmon/hwmon6# cat temp1_input
30908
root@rockos-eswin:/sys/class/hwmon/hwmon6#
```

I'm not sure how accurate it is, other than to say, by touch, the Heatsink doesn't not even get warm, so i'm going to go with a "its pretty accurate" 

Let me know if you are interested in anything else:
```
root@rockos-eswin:/sys/class/hwmon# cat /sys/class/hwmon/hwmon?/label
npu_vdd
cpu_vdd
soc_vdd
lpddr_vdd
dc_vdd
fan_control
pvt0
pvt1
root@rockos-eswin:/sys/class/hwmon#```

Fishwaldo, 2024-08-25:
(pvt1 is the NPU temp/voltages)

Fishwaldo, 2024-08-25:
I've also been told I can flip a dip switch to get it to clock at 1.8Ghz... its just that I am still pending hardware documents/schematics. When I get that, I'll re-run the test if it is true that we can change the max frequency.

ThomasKaiser, 2024-08-26:
> maybe its just the BSP kernel drivers?)

Sure, they should try to let these thermal readouts not only appear below `/sys/class/hwmon` but also as `/sys/devices/virtual/thermal/thermal_zoneN`, at least for the SoC/CPU temperature labeled as `cpu-thermal` _or_  `soc-thermal`.

Looking through `sbc-bench` results it's quite a variety of node names:

    tk@mac-tk results % grep "^Thermal source" *.txt | awk -F" " '{print $4}' | sort | uniq -c | sort -n -r
      38 (cpu-thermal)
      31 (cpu_thermal)
      21 (soc-thermal)
      14 (x86_pkg_temp)
       9 (soc_thermal)
       9 (cpu0-thermal)
       6 (cpu)
       4 (soc)
       4 (cpu_thermal_zone)
       4 (SOC2-therm)
       3 (cpu6-thermal)
       3 (cpu-thermal-zone)
       2 (soc_dts0)
       2 (scpi_sensors)
       2 (cpu7-thermal)
       2 (cpu0_thermal)
       1 (thsens_cpu)
       1 (soc-thmzone)
       1 (imx_thermal_zone)
       1 (iio_hwmon)
       1 (f10e4078.thermal)
       1 (cpu_little4)
       1 (cpu-1-2-usr)
       1 (ap-thermal-cpu3)
       1 (acpitz)
       1 (CPU-therm)
       1 (120e0000.tmon)

> hwmon6 is the CPU temp

May I still ask for `/sys/class/hwmon/hwmon6/name` since then it's just adjusting a `grep` pattern for now to let the CPU temperature been included in your next `sbc-bench` run at 1.8 GHz (which I prefer to include into results lists since if the HW is capable of 1.8 GHz this should be reflected by results).

Once I got `/sys/class/hwmon/hwmon6/name` I will push a small commit fixing SoC detection and thermals, so after an `sbc-bench -u` you can retest at 1.8 GHz :)

ThomasKaiser, 2024-08-28:
Closed by https://github.com/ThomasKaiser/sbc-bench/commit/76420fff0913e7626224879291295b1d63dc4fcd

closed 2024-08-28 by ThomasKaiser

Fishwaldo, 2024-08-29:
Hi ,
here is the name of the sysfs node:

```
root@rockos-eswin:/sys/class/hwmon# more /sys/class/hwmon/hwmon6/name
pvt
```

very non-descriptive!

ThomasKaiser, 2024-08-29:
> very non-descriptive!

Yeah :) 

But since the only other `hwmon` node labeled the same is in TH1520's BSP kernel where CPU temp is also exported as `/sys/devices/virtual/thermal/thermal_zone0/` https://github.com/ThomasKaiser/sbc-bench/commit/236f4a05169958e58c8da948e14cd44dd8abc640 should now give thermal readouts on this platform w/o harming others.

As such after an `sbc-bench -u` you're ready for retesting at 1.8 GHz :)