ThomasKaiser/sbc-bench #102: Sipeed Lichee Pi 3A

geerlingguy, 2024-09-05:
```

ThomasKaiser, 2024-09-06:
Interestingly according to  `/sys/devices/soc0/` this board features a 'K1-6370 rev C' vs. ['M1-8571 rev C' on your Milk-V Jupiter](https://github.com/ThomasKaiser/sbc-bench/blob/master/results/X904.txt#L511). Not sure though from which source `/sys/devices/soc0/` gets populated. If it's DT then it's just settings and not hardware.

Ah, it's indeed hardware revisions [since the SoC's serial number gets parsed](https://github.com/TeddyNight/linux-6.1/blob/b6f0c194ef8caf00902a118abb6133fe585d10c8/drivers/soc/spacemit/spacemit-socinfo.c#L39-L45).

> It doesn't seem like the result was uploaded though...?

Happens from time to time. Uploading results will eve be tried twice but when the hoster in question (after https://ix.io and https://sprunge.us becoming disfunctional the script currently tries https://0x0.st) is unavailable for a moment uploading full results will fail.

In case time permits you could rerun `sbc-bench` or simply try to upload contents of `/var/log/sbc-bench.log`:

    sed '/^$/N;/^\n$/D' </var/log/sbc-bench.log  | sed "s,\x1B\[[0-9;]*[a-zA-Z],,g" | curl -s -F 'file=@-' 0x0.st

geerlingguy, 2024-09-10:
Results now here: https://0x0.st/XxZl.bin

This thing's performance is... interesting. Definitely a step down from the 'M1' version with a higher clock, but not that much.

commit 84e251d1 (2024-09-10): Adding Lichee Pi 3A scores and Lunar Lake support
Closes #102
closed 2024-09-10 by ThomasKaiser