ThomasKaiser/sbc-bench #105: AmpereOne M192-32 192-core server

geerlingguy, 2024-10-25:
It's, uh... not quite an SBC. But it is Arm!

https://0x0.st/XUpQ.bin

ThomasKaiser, 2024-10-26:
Thanks a lot, this is an interesting system using custom ARM cores, showing excellent RAM latency and [easily outperforming the former 'champion' wrt 7-ZIP MIPS](https://github.com/ThomasKaiser/sbc-bench/blob/47029a8f5d8d7c30bc9b47abb76709e0af898fe7/Sorted-Results.md#7-zip-mips-multi-threaded). :)

closed 2024-10-26 by ThomasKaiser

geerlingguy, 2024-10-27:
Indeed, this may be the first not-cloud-only Arm server available with DDR5 RAM? This one set at 5200 MT/sec, probably can go higher though?

geerlingguy, 2024-10-29:
@ThomasKaiser - New results: https://0x0.st/X0gc.bin

On the previous run, one of the DIMMs installed was throwing tons of errors, so the new result may have a few numbers a bit better.

ThomasKaiser, 2024-10-30:
> the new result may have a few numbers a bit better.

Thanks for retrying though I don't see an indication for memory performance having improved. Looks just like results variation since the ECC errors all were single bit flips that got corrected on-the-fly at the memory controller layer.

> Bumped 7zip multi from 741530 to 745720

No, this was results variation as well and the lower 1st average score was due to the first of the three 7-zip runs being lower for whatever reasons:

  * 7-zip MIPS (3 consecutive runs): **738751**, 741592, 744259 (741530 avg), single-threaded: 4781

vs. this later

  * 7-zip MIPS (3 consecutive runs): 743136, 748159, 745863 (745720 avg), single-threaded: 4783

commit 47029a8f (2024-10-26): Adding AmpereOne A192-32X scores
Closes #105