ThomasKaiser/sbc-bench #107: [priority none] question about experience on NTFS difficulty with Linux git repositories

gitthnx, 2025-01-03:
Did you ever experience or hear about (at least contemporary) incompatibilities for git repositories ([e.g. u-boot](https://source.denx.de/u-boot/u-boot/-/issues/44#note_188105)) not being able to compile/build on NTFS filesystems, but nicely compiling (same repository cloned, same OS distribution&hw) on e.g. ext4? 
( You are requesting attention for exceptional behavior on NTFS https://github.com/ThomasKaiser/sbc-bench#ntfs-filesystems, maybe You can add some experience to that (and maybe it would be a more common difficulty if more people would utilize NTFS for storage for git repositories and build from there)?
Maybe this would happen more likely on x86_64 (SBC) systems that have Windows filesystems (NTFS) on storage devices, that arm64/riscV devices not capable of running Windows OS? )
(thx)

ThomasKaiser, 2025-01-03:
> Did you ever experience or hear about (at least temporarily) incompatibilities for git repositories (e.g. u-boot) not being able to compile/build on NTFS filesystems

Nope. Not using NTFS for decades (last century we had to use some Windows servers due to lack of software alternatives) I lack practically any experience. The only thing I know for sure is NTFS on Linux being a bad choice especially when being mounted with FUSE. There are many _native_ filesystems available on Linux (ZFS, btrfs, XFS, ext4, and so on) so IMHO there's no point dealing with the hassles of a foreign filesystem.

closed 2025-01-03 by ThomasKaiser

gitthnx, 2025-01-03:
Thanks for Your response and clarifications (&best wishes for changing years)

for notice (only): 
From my experience, NTFS is 'slower' on Linux OS for searching, copying, moving and deleting directories with 'many' files in it (like repositories), but it was the least used partition on a shared disk (multi boot with Windows OS included and access from Windows OS required) and NTFS would (almost) never be a recommendation/option for a commonly implemented storage for Linux OS. 
At the moment i cannot test sbc-bench.sh git cloning on that NTFS filesystem, because it's delayed for waiting for a lower utilized system. Maybe i can add some experience with having this feature skipped for testing repository building on NTFS, just for own interest about that rare issue.(WIP_03012025)
(thx)