ThomasKaiser/sbc-bench #110: Power consumption metrics

ValdikSS, 2025-02-04:
Hey tkaiser,

While the CPU performance tests could be found for different SBCs here and there (and in your great table of course!), power consumption tests are usually omitted, and it's not easy to find it even for relative comparison between different architectures/chip families.

For example, my Rockchip RK3308 on Raxda Rock S consumes 0.05W suspended, 0.2W idle, 1.5W under full load (all 4 cores), while RK3588 on Orange Pi 5+, according to [Armbian forum post](https://forum.armbian.com/topic/48383-power-usage-with-latest-kernel-612-server-ubuntu-build/), used to consume 1.5W idle minimum, and 5W idle on the latest kernel.

Even the inaccurate numbers would allow to understand whether this SoC is suitable for a battery-powered device, does it properly power off and has working suspend, and how's overall power situation.

Ideal tests I'd like to see:

* Power consumption in `poweroff` from the system
* Power consumption in suspend
* Power consumption in [standby](https://www.kernel.org/doc/html/latest/admin-guide/pm/sleep-states.html#standby)
* Power consumption in idle, Wi-Fi/Ethernet disabled
* Power consumption in idle, with Wi-Fi/Ethernet individually measured

Power numbers could be recorded from the USB power tester. There are two known-good popular models: Fnirsi FNB58 and AVHzY CT3 (Shizuku). Don't know much about the former, but the latter (which I have) not only automatically collects charge data and shows real-time values on the screen, but also allows LUA scripting.

Please tell me what do you think. If you're happy with this idea, I could begin the contribution with 4 SBC tests I own.

ThomasKaiser, 2025-02-04:
> does it properly power off and has working suspend, and how's overall power situation.

All of these are also software things and just like users don't understand that you can't benchmark hardware (since you're always benchmarking both hardware and software/settings) they won't understand this here as well. As such I personally gave up on this or just in the form of (p)reviews where every number provided is accompanied by the disclaimer "this may or will change for sure in the future".

Some examples:

  * when I measure Orion O6 now `poweroff` results in 4W drawn at the wall and `idle` is north of 13W depending on how you measure and which kernel modules you blacklist. Are the numbers relevant? Not at all since they will change or have to change for Radxa/Cix not loosing their face
  * every RPi so far on day one showed higher consumption and lower performance than later since constant firmware updates adjust this and that
  * on other SBC the usual situation is as follows: the boards start shipping with a crappy BSP kernel everyone hates (but which comes with settings/drivers that allow the Android devices these SoCs were designed for in the first place to be at least somewhat energy efficient). Once community starts to take over and the devices run with 'latest and greatest' mainline kernel not only consumption usually increases (often by a lot, see RK3588 situation) but often also performance decreases. Real-world performance not stupid benchmarks that run the devcice full speed

How should a consumer deal with the '1.5W in idle' number for a RK3588 devices when in reality after next update then running with mainline the thing needs 5W in reality (due to kernel hackers usually picking up the interesting stuff and power rails and such stuff being not amongst this).

Why publishing 5W/15W `poweroff`/`idle` numbers for an O6 for example when it's obvious that this will soon change since software (both various firmwares on MCUs inside the SoC as well as kernel/drivers for the ARM cores) is not ready yet?

Please don't get me wrong: the idea is great but just collecting a bunch of numbers somewhere in a table doesn't help anyone. Collecting/publishing those numbers is something for responsible reviewers that are able to garnish numbers with disclaimers based on prior experience.

For power measurements I added a NetIO mode ([switchable power meters that measure at the wall](https://www.netio-products.com/en)), see for example here the most right column: https://github.com/ThomasKaiser/sbc-bench/blob/8220da31b36a05ae341d7a31acc9de01c479a8c3/results/888t.txt#L1658-L1751

 Am planning to add support for ODROID SmartPower 3 now for almost a year to measure between PSU and board but got distracted by other stuff. But at least the power meter in question should have an API that allows it to be queried while `sbc-bench` is running since only this allows for exact (that means averaged) idle measurements as well as iterating through specific policies/governors, see for example at the very end of this O6 check: https://github.com/ThomasKaiser/sbc-bench/blob/8220da31b36a05ae341d7a31acc9de01c479a8c3/results/888t.txt#L2099-L2108 (though the label is wrong, it should read `powersave` instead of `performance`)

ValdikSS, 2025-02-04:
Sure, that's why I want to check software+hardware combinations. Ideally, a SBC manufacturer-provided image and third-party image like Armbian, to compare them, maybe several versions, but the latest should be good enough, assuming that manufacturer image/BSP kernel is a "golden standard" from their point of view.

This represents the current state of the board support, and that's exactly what I would like to know. Not a "plain" hardware capabilities, but a realistic representation of what you can get on this or that SBC/SoC.

I care more about lower-end chips, which are usually present in SBCs which are at least several years old, sans the SoCs itself that are older, so it should be fairly representative. Don't know how to deal with fresh chips for which everything may change rapidly.