ThomasKaiser/sbc-bench #112: Phytium D3000

ThomasKaiser, 2025-02-19:
@ neggles may I ask for an `sbc-bench -r` run on your D3000 mATX CRB? Would be great to have results for the FTC862 cores in the list :)

neggles, 2025-04-28:
oh hey yeah sure! (sorry I totally missed this notification, need to clean up my GH notifs sometime...)

I don't know that it really qualifies as an SBC per se, mind you... 😝

neggles, 2025-05-05:
here we go: [0x0.st/84GG.bin](https://0x0.st/84GG.bin)
also attached: [84GG.txt](https://github.com/user-attachments/files/20037997/84GG.txt)

closed 2025-05-05 by ThomasKaiser

ThomasKaiser, 2025-05-05:
That's quite a performant thing :)

Unlike the previous Phytium cores that looked like Cortex-A57 derivates the FTC862 here show an ARMv8 Crypto Extensions implementation similar to more recent 'big' cores: https://github.com/ThomasKaiser/sbc-bench/blob/master/results/ARMv8-Crypto-Extensions.md

Well, it's ARMv8.2 vs. v8.0 :)

commit d3af83c5 (2025-05-05): Adding Phytium D3000/8 scores
Closes #112