ThomasKaiser/sbc-bench #113: Test run of Snapdragon Dev Kit X1E001DE

jglathe, 2025-03-08:
| Qualcomm Technologies, Inc. X1E001DE Snapdragon Devkit for Windows | ~3790 | 6.14 | Ubuntu 24.10 (oracular) arm64 | 66740 | 5526 | 2273920 | 35880 | 67810 | - | https://0x0.st/8uEs.txt |

ThomasKaiser, 2025-03-10:
Thank you for the submission!

But may I ask you to update `sbc-bench` and retest. In https://github.com/ThomasKaiser/sbc-bench/commit/f4858d1067e7a1e31aebbcaab26536f3b5da30b8 I tried to detect the two CPU clusters better or at all.

The problem is Qualcomm again somehow creatively using revision/variant to distinguish different CPU cores that share the same CPU implementer/part numbers (they already did that a decade ago with some of their Krait lineup).

Updating is simple: `sbc-bench -u` followed by another run that then indivividually tests also `cpu4` and `cpu0-cpu3` vs. `cpu4-cpu11` :)

jglathe, 2025-03-11:
Hi @ThomasKaiser , these are the values from the new run:

| Qualcomm Technologies, Inc. X1E001DE Snapdragon Devkit for Windows | ~4130 | 6.14 | Ubuntu 24.10 (oracular) arm64 | 65320 | 6110 | 2215310 | 35650 | 100220 | 75.49 | https://0x0.st/8SAz.txt |

ThomasKaiser, 2025-03-11:
Thanks a bunch @jglathe!

Though the 2nd cluster still being limited to 4130 MHz instead of the advertised 4300 MHz [I gave it another try to consider this three clusters instead of two](https://github.com/ThomasKaiser/sbc-bench/commit/eff3b54e3a4e918c4711c0d0c1c61ce2b460bd09#diff-fdfd2a032c64d6e9ba92a3197cad6b26573c7094433d74efa4ae80f44f65aa99). So in case time permits please re-test again :)

jglathe, 2025-03-11:
Hi @ThomasKaiser , this is the result... 

```
jglathe@angrybox:~/src/sbc-bench$ sudo ./sbc-bench.sh -c
WARNING: This tool is meant to run only on recent Debian, Fedora, Ubuntu or Rocky Linux distros.

When executed on Ubuntu 24.10 (oracular) results might be partially meaningless.
Please see https://github.com/ThomasKaiser/sbc-bench/issues/81 for reasons.

Press [ctrl]-[c] to stop or [enter] to continue.

Average load and/or CPU utilization too high (too much background activity). Waiting...

Too busy for benchmarking: 16:48:21 up  7:06,  2 users,  load average: 1.02, 1.44, 1.58,  cpu: 5%
Too busy for benchmarking: 16:48:27 up  7:06,  2 users,  load average: 0.94, 1.41, 1.57,  cpu: 1%
Too busy for benchmarking: 16:48:32 up  7:06,  2 users,  load average: 0.95, 1.40, 1.57,  cpu: 1%
Too busy for benchmarking: 16:48:37 up  7:06,  2 users,  load average: 0.87, 1.38, 1.56,  cpu: 0%
Too busy for benchmarking: 16:48:42 up  7:06,  2 users,  load average: 0.80, 1.36, 1.55,  cpu: 0%
Too busy for benchmarking: 16:48:47 up  7:06,  2 users,  load average: 0.74, 1.34, 1.54,  cpu: 0%

Status of performance related governors found below /sys (w/o cpufreq):
3d00000.gpu: simple_ondemand / 300 MHz (userspace powersave performance simple_ondemand / 300 390 550 687 744 800 925 1000 1100)

sbc-bench v0.9.71

Installing needed tools: distro packages already installed. Done.
Checking cpufreq OPP. Done (results will be available in 20-30 minutes).
Executing tinymembench. Done.
Executing RAM latency tester. Done.
Executing OpenSSL benchmark. Done.
Executing 7-zip benchmark. Done.
Executing cpuminer. 5 more minutes to wait. Done.
Checking cpufreq OPP again. Done (20 minutes elapsed).

Results validation:

  * Advertised vs. measured max CPU clockspeed: -1.4% before, -1.4% after -> https://tinyurl.com/32w9rr94
  * No swapping
  * Background activity (%system) OK
  * No throttling
  * 3 different clusters but capacity-dmips-mhz property not set
  * schedutil cpufreq governor configured but neither dynamic-power-coefficient nor sched-energy-costs defined

Memory performance (all 3 CPU clusters measured individually):
memcpy: 36500.2 MB/s (Qualcomm Oryon-X1)
memset: 72067.1 MB/s (Qualcomm Oryon-X1)
memcpy: 37547.9 MB/s (Qualcomm Oryon-X1)
memset: 108554.4 MB/s (Qualcomm Oryon-X1)
memcpy: 38363.2 MB/s (Qualcomm Oryon-X1)
memset: 110408.3 MB/s (Qualcomm Oryon-X1)

Cpuminer total scores (5 minutes execution): 76.89,76.85,76.84,76.80,76.69,76.57,76.54,76.47,76.39,76.29,76.28,76.19,76.17,76.11,76.07,76.04,76.00,75.90,75.87,75.84,75.83,75.80,75.71,75.63,75.62,75.60,75.56,75.48,75.46,75.43,75.42,75.41,75.36,75.35,75.34,75.30,75.24,75.22,75.20,75.19,75.18,75.16,75.15,75.14,75.13,75.07,75.03,74.99,74.95,74.91 kH/s

7-zip total scores (3 consecutive runs): 68221,64086,66040, single-threaded: 5603

OpenSSL results (all 3 CPU clusters measured individually):
type             16 bytes     64 bytes    256 bytes   1024 bytes   8192 bytes  16384 bytes
aes-128-cbc    1703843.72k  2289748.14k  2439264.77k  2492901.72k  2511937.54k  2514796.54k (Qualcomm Oryon-X1)
aes-128-cbc    1761508.12k  2567667.48k  2757959.51k  2803996.33k  2790517.42k  2822056.62k (Qualcomm Oryon-X1)
aes-128-cbc    1915843.70k  2585884.84k  2757943.13k  2819723.61k  2820030.46k  2808769.19k (Qualcomm Oryon-X1)
aes-192-cbc    1592278.42k  1968807.45k  2156421.12k  2201873.07k  2230910.98k  2227732.48k (Qualcomm Oryon-X1)
aes-192-cbc    1654998.01k  2220309.53k  2452034.22k  2508852.22k  2526052.35k  2546537.81k (Qualcomm Oryon-X1)
aes-192-cbc    1781649.74k  2219717.33k  2450084.18k  2505335.47k  2522374.14k  2524069.89k (Qualcomm Oryon-X1)
aes-256-cbc    1384070.49k  1766685.06k  1902652.25k  1937759.57k  2014855.17k  2015264.77k (Qualcomm Oryon-X1)
aes-256-cbc    1480346.77k  1994969.00k  2149596.07k  2188664.49k  2276002.47k  2274186.58k (Qualcomm Oryon-X1)
aes-256-cbc    1546293.37k  1976306.52k  2123715.58k  2146073.94k  2254310.06k  2271794.52k (Qualcomm Oryon-X1)

Full results uploaded to https://0x0.st/8SWL.txt

```

jglathe, 2025-03-11:
It has pushed the box a little more IMO. Fan was audible, but not as much as on a kernel build run.

closed 2025-03-12 by ThomasKaiser

ThomasKaiser, 2025-03-12:
Thanks for re-testing.

The result variation is quite high (I would assume the SoC's firmware also implementing power limits) but at least it seems both the 2nd and 3rd cluster can clock at up to 4300 MHz. And of course this thing is a beast! :)

jglathe, 2025-03-12:
For the third test I have killed Firefox, so this may be a partial cause. Otherwise, yes its powerful for such a little box. Build-wise it equals the 30 threads VM on a Ryzen 9 9350. Which draws more than 200W, though 😉

ThomasKaiser, 2025-03-12:
> For the third test I have killed Firefox, so this may be a partial cause

Indeed, looking at overall CPU utilization with the single-threaded benchmarks, there's close to 1% difference in total utilization.

Asides that the actual measuring of clockspeeds on recent Ubuntus needs an update but since you always updated `sbc-bench` with the `-u` switch, the `ramlat` tool has been rebuilt if necessary. I owe @wtarreau an answer about his latest findings but currently too busy with `$dayjob` and interests have shifted within the last year (less time in front of displays and more with [physical tools](https://github.com/ThomasKaiser/Basteln/tree/main/media/Anhaengerdeckel#readme))

wtarreau, 2025-03-13:
@ThomasKaiser no worries, I just sent you that stuff as a followup of an old conversation, I'm not waiting for an answer and am, too, super busy with $dayjob. BTW the beast above is impressive. I don't know how large the memory area is for the RAM tests because that speed suggests at least 4 channels at 8000 MT/s, Or maybe the area is too narrow and the device has a huge cache.

One test which would be worth adding and that reflects integer performance quite well is the openssl rsa speed test (e.g. rsa2048). The algorithms have already been super-optimized to try to get the best on modern processors, and the code is no longer progressing. That's an area where you easily see an x2 difference between performance and efficiency cores. The only annoying thing is that the test runs for 10s per pass, and that recent openssl versions have changed the output format a bit and run more tests.

jglathe, 2025-03-13:
It is 32GB RAM, 8448MT

```
jglathe@angrybox:~$ sudo lshw -short
H/W path     Device           Class      Description
====================================================
                              system     Qualcomm Technologies, Inc. X1E001DE Snapdragon Devkit for Windows
/0                            bus        Motherboard
/0/1                          memory     1MiB BIOS
/0/2                          memory     2304KiB L1 cache
/0/3                          memory     1152KiB L1 cache
/0/4                          memory     36MiB L2 cache
/0/6                          memory     32GiB System Memory
/0/6/0                        memory     32GiB TSOP 8448 MHz (0.1 ns)
/0/9                          processor  ARM (446)
/0/5                          processor  cpu-map
/0/7                          processor  cpu
/0/8                          processor  cpu
/0/a                          processor  cpu
/0/b                          processor  cpu
/0/c                          processor  cpu
/0/d                          processor  cpu
/0/e                          processor  cpu
/0/f                          processor  cpu
/0/10                         processor  cpu
/0/11                         processor  cpu
/0/12                         processor  cpu
/0/13                         processor  cpu
/0/14                         processor  domain-idle-states
/0/15                         processor  idle-states
/0/100                        bridge     Qualcomm Technologies, Inc
/0/100/0     wlP4p1s0         network    WCN785x Wi-Fi 7(802.11be) 320MHz 2x2 [FastConnect 7800]
/0/101                        bridge     Qualcomm Technologies, Inc
/0/101/0     enP5p1s0         network    RTL8125 2.5GbE Controller
/0/0                          bridge     Qualcomm Technologies, Inc
/0/0/0       /dev/nvme0       storage    KIOXIA-EXCERIA PLUS G3 SSD
/0/0/0/0     hwmon55          disk       NVMe disk
/0/0/0/2     /dev/ng0n1       disk       NVMe disk
/0/0/0/1     /dev/nvme0n1     disk       2TB NVMe disk
/0/0/0/1/1   /dev/nvme0n1p1   volume     1023KiB EFI partition
/0/0/0/1/2   /dev/nvme0n1p2   volume     15EiB Windows FAT volume
/0/0/0/1/3   /dev/nvme0n1p3   volume     1023KiB EFI partition
/0/0/0/1/4   /dev/nvme0n1p4   volume     1023KiB EFI partition
/0/0/0/1/5   /dev/nvme0n1p5   volume     69MiB EFI partition
/0/0/0/1/6   /dev/nvme0n1p6   volume     119MiB EFI partition
/0/0/0/1/7   /dev/nvme0n1p7   volume     119MiB EFI partition
/0/0/0/1/8   /dev/nvme0n1p8   volume     1023KiB EFI partition
/0/0/0/1/9   /dev/nvme0n1p9   volume     1023KiB EFI partition
/0/0/0/1/a   /dev/nvme0n1p10  volume     1023KiB EFI partition
/0/0/0/1/b   /dev/nvme0n1p11  volume     511MiB EFI partition
/0/0/0/1/c   /dev/nvme0n1p12  volume     15MiB reserved partition
/0/0/0/1/d   /dev/nvme0n1p13  volume     519MiB Windows FAT volume
/0/0/0/1/e   /dev/nvme0n1p14  volume     999MiB Windows NTFS volume
/0/0/0/1/f   /dev/nvme0n1p15  volume     474GiB Windows NTFS volume
/0/0/0/1/10  /dev/nvme0n1p16  volume     1386GiB EXT4 volume
/1           usb1             bus        xHCI Host Controller
/1/2                          bus        USB2.0 Hub
/1/2/1       input0           input      Cherry GmbH CHERRY Corded Device
/1/2/2       input2           input      ROCCAT ROCCAT Kone Aimo 16K
/1/2/3                        bus        USB2.0 Hub
/2           usb2             bus        xHCI Host Controller
/2/2                          bus        USB3.0 Hub
/2/2/3                        bus        USB3.0 Hub
/3           mmc0             bus        MMC Host
/4           /dev/fb0         display    msmdrmfb
/5           input7           input      pmic_pwrkey
```

commit e6d4fd62 (2025-03-12): Final results for X1E001DE
Closes #113