ThomasKaiser/sbc-bench #115: Orion O6 - 2nd round with SystemReady Firmware

geerlingguy, 2025-04-27:
I notice you have the preliminary results up, but here's a 2nd round of testing: https://0x0.st/8WAL.bin

Note that some things like temps in various places didn't seem to report correctly, and for example I got the following message:

```
In case this device is not already represented in official sbc-bench results list then please
consider submitting it at https://github.com/ThomasKaiser/sbc-bench/issues with this line:
 https://0x0.st/8WAL.bin |

```

ThomasKaiser, 2025-04-28:
Thank you though I don't think using the SystemReady 'firmware' is of any use for us Linuxers :)

As Yuntian explained they had to publish this UEFI release due to the certification process but are still preparing another release that is meant for us.

As for the results. `sbc-bench` is lacking support/hacks for this weird 'P1 in SystemReady' mode and I don't plan to add them (still waiting for both an OS image and UEFI to be targeted at Linux users and not ARM certifications). The CPU cores in ACPI mode are presented as this:

    CPU sysfs topology (clusters, cpufreq members, clockspeeds)
                     cpufreq   min    max
     CPU    cluster  policy   speed  speed   core type
      0        0        0      800    2600   Cortex-A720 / r0p1
      1        0        1      800    2300   Cortex-A720 / r0p1
      2        0        1      800    2300   Cortex-A720 / r0p1
      3        0        3      800    2200   Cortex-A720 / r0p1
      4        0        3      800    2200   Cortex-A720 / r0p1
      5        0        5      800    2500   Cortex-A720 / r0p1
      6        0        5      800    2500   Cortex-A720 / r0p1
      7        0        5      800    2500   Cortex-A720 / r0p1

Which is essentially the same as we had in January since back then `cpu7` (which is `cpu11` when the little cores are also present) was also listed at 2.5 GHz but was running with 2.6 GHz too. I hacked cluster detection together back then and all 6 'clusters' (the fastest last one being artificially moved to `cpu0` for whatever reasons being handled as own cluster) were measured individually: https://github.com/ThomasKaiser/sbc-bench/blob/master/results/88LE.txt#L1148-L1153

`cpu0` scored 1458416.30k and `cpu11` 1458410.84k with the openssl benchmark which means both ran at 2.6 GHz. For reasons see https://github.com/ThomasKaiser/sbc-bench/blob/master/results/ARMv8-Crypto-Extensions.md (the TL;DR is: openssl scores are not affected by RAM performance and when the core type is known and ARM Crypto Extensions present the clockspeed can be calculated from the score).

Now with your run obviously the scheduler has no clue to run which task on which core. In case `sbc-bench` detects a system with all cores of same type it handles this as a single cluster and does not try to test through same CPU cores of different clockspeeds. In this mode the OpenSSL benchmark will be executed two times and based on your results the scheduler executed the first run on `cpu0` or `cpu7` (2.6 GHz) and the 2nd run on `cpu5` or `cpu6` (2.5 GHz):

    aes-256-cbc     805333.15k  1295310.04k  1421821.10k  1449060.35k  1456283.65k  1456750.59k
    aes-256-cbc     777065.52k  1245645.80k  1367112.96k  1393771.52k  1400086.53k  1400487.94k

(1456751 / 561 = 2.596 MHz, 1400488 / 561 = 2.496 MHz)

The OpenSSL benchmark in `sbc-bench` is somewhat special since when only a single CPU cluster is 'detected' as in this 'Cix P1 with SystemReady firmware' case then `openssl speed -elapsed -evp aes-${bytelength}-cbc` is executed twice but w/o pinning the execution to a specific core (just as Geekbench is doing it). And as we see with this UEFI variant they may then end up on a core allowed to clock at 2.6 GHz or only 2.5 GHz. And maybe further testing might reveal that 2.2/2.3 GHz are also an option. A simple test accompanied by a 'taskset -c 3 htop` in another shell would be

    for i in $(seq 1 10) ; do openssl speed -elapsed -evp aes-256-cbc 2>&1 | grep "^AES-256-CBC" ; done

For now we only know that with UEFI version 9.0.0 the scheduler might send power hungry tasks to cores at either 2.6 or 2.5 GHz. But in case you're testing we may see that 2.2 and 2.3 GHz are also an option. Benchmarking such a system is therefore partially using a random number generator :)

geerlingguy, 2025-04-28:
That may also explain the abysmal HPL score I got, even slower than the already slow one I had on the older firmware.

ThomasKaiser, 2025-04-28:
> That may also explain the abysmal HPL score I got

I always thought HPL runs massively parallel as in 'on all CPU cores simultaneously'? If so then scheduler decisions what to run where shouldn't matter that much since it should run on all cores. :)

Seriously no idea: I've only looked into Linpack combined with OpenBLAS as a stability tester ([automated testing through DVFS settings](https://github.com/ThomasKaiser/StabilityTester)) after ejolson [1] discovered DVFS related instabilities with RPi 3B or 3B+ using this approach. He constantly explained over there that HPL is not a 'fire and forget' benchmark (more than one time to Gareth Halfacree ;) ) but that's almost all I know about this benchmark since it doesn't represent anything I do with computers.

But if your O6 scores are lower than those generated with a BCM2712 then there must be something wrong :)

[1] https://forums.raspberrypi.com/memberlist.php?mode=viewprofile&u=104329

geerlingguy, 2025-04-28:
Oh I know—the thing is, most vendors have a keen interest in making sure any extensions their chips are using are well supported in OpenBLAS, BLIS, etc. as those math libraries seem to be used in a lot of HPC and even in some smaller distributed computing benchmarks.

But other Armv8 and Armv9 chips are well-supported, and I've run it on systems with all-performance, all-efficiency, and mixed big.LITTLE cores and never had a *huge* surprise like with the Cix chip.

If you can get micro-optimizations, like Oracle's version of BLIS for Ampere chips, you can get a 10-20% performance/efficiency increase. But I've never seen such a poor result on any chip before, which has me wondering if there's some memory access weirdness when you run it across all cores. The power consumption during the HPL run indicates it's utilizing all the horsepower at least. It's just... showing a terrible end result.

I may play with the Ps and Qs a bit more to see if there's some more optimal run configuration.

ThomasKaiser, 2025-04-28:
Can't it be something as trivial as 'no detection for ARMv9 as such running w/o any optimizations'?

(just a thought and hopefully as trivial to be ruled out :) )