ThomasKaiser/sbc-bench #116: Results:  Thundercomm Rubik Pi C6490P (Qualcomm QCS6490)

swdee, 2025-04-28:
| Thundercomm, Inc. Rubik Pi C6490P | ~2400 | 6.6 | Debian GNU/Linux trixie/sid arm64 | 14910 | 3547 | 1355120 | 7030 | 17270 | 19.76 | https://0x0.st/8WQX.txt |

ThomasKaiser, 2025-04-28:
Thank you! Just wondering why `sbc-bench` skipped benchmarking the fastest core `cpu7`. Can you please provide output of these two commands?

    ls -ld /sys/devices/system/cpu/cpufreq/policy?
    cat /sys/devices/system/cpu/cpu*/topology/physical_package_id

swdee, 2025-04-28:
```
$ ls -ld /sys/devices/system/cpu/cpufreq/policy
ls: cannot access '/sys/devices/system/cpu/cpufreq/policy': No such file or directory
```

```
$ cat /sys/devices/system/cpu/cpu*/topology/physical_package_id
0
0
0
0
0
0
0
0
```

The available cpufreq/policy files are;
```
$  ls -la  /sys/devices/system/cpu/cpufreq/
total 0
drwxr-xr-x  5 root root 0 Jan  1  1970 .
drwxr-xr-x 16 root root 0 Jan  1  1970 ..
drwxr-xr-x  4 root root 0 Jan  1  1970 policy0
drwxr-xr-x  4 root root 0 Jan  1  1970 policy4
drwxr-xr-x  4 root root 0 Jan  1  1970 policy7
```

swdee, 2025-04-29:
Withdraw the original results posted as it was pointed out to me that thermal throttling had occurred.   I added the Raspberry Pi 5 heatsink/fan and ran the bench again with these new results.

| Thundercomm, Inc. Rubik Pi C6490P | ~2400 | 6.6 | Debian GNU/Linux trixie/sid arm64 | 16190 | 3536 | 1356100 | 7040 | 17070 | 23.34 | https://0x0.st/8Wvr.txt |

swdee, 2025-04-29:
Third attempt, this time we achieved a result with thermal throttling.

| Thundercomm, Inc. Rubik Pi C6490P | ~2400 | 6.6 | Debian GNU/Linux trixie/sid arm64 | 17830 | 3535 | 1364250 | 7020 | 17000 | 27.50 | https://0x0.st/8Wxo.txt |

ThomasKaiser, 2025-04-29:
Great, though if time permits can you please run this 'hacked' version here: https://gist.githubusercontent.com/ThomasKaiser/e30acf8b31e1b94a69fe50348e94fc4b/raw/c5209e57c7d3e58c63cb5e9cf4c124f66e777188/gistfile1.txt

I've still no clue why the cluster detection fails so it's hardcoded in this variant. This will then test through `cpu0-cpu3`, `cpu4-cpu6` and also `cpu7` which should result in properly reported single-threaded scores :)

swdee, 2025-04-29:
Here are the results for the hacked version;

| Thundercomm, Inc. Rubik Pi C6490P | 2707/1958 MHz | 6.6 | Debian GNU/Linux trixie/sid arm64 | 17920 | 3871 | 1539030 | 7020 | 17230 | 27.50 | https://0x0.st/8WlQ.txt |

As for the cluster detection I did a little debugging.

Modifying the `GetCoreClusters()` function to echo its output;
```
GetCoreClusters() {
        # function to determine clusters by CPU core type used by MODE=extensive
        # and Geekbench mode
        #
        # With x86 we rely on name based cluster detection while on ARM and other
        # platforms we 'bundle' CPU core types by occurence:
        #
        # Amlogic S912 for example contains 2 quad-core A53 clusters with different
        # cpufreq scaling properties but it's just 8 boring A53 and no big.LITTLE
        # so there's no reason to treat S912 as '2 clusters CPU'.
        #
        # RK3588 for example consists of 4 x A55 cores and 4 x A76 but the latter
        # are handled as two different clusters sharing same properties except
        # clockspeed and throttling differences for a reason called PVTM:
        # https://github.com/ThomasKaiser/Knowledge/blob/master/articles/Quick_Preview_of_ROCK_5B.md#pvtm

        if [ "${CPUArchitecture}" = "x86_64" ]; then
                Getx86ClusterDetails
        else
                local i

echo "CPU Cores = ${CPUCores[@]}" > /dev/tty
                for i in $(seq 0 $(( ${CPUCores} - 1 )) ) ; do
                        ThisCore="$(GetCPUInfo $i)"
echo "This Core i=${i} = ${ThisCore[@]}" > /dev/tty
                        if [ "X${ThisCore}" != "X${LastCore}" ]; then
                                echo "${i}"
                                LastCore="${ThisCore}"
echo "inside if i=${i}, LastCore=${LastCore[@]}" > /dev/tty
                        fi
                done
        fi
} # GetCoreClusters
```

This results in output of;
```
CPU Cores = 8
This Core i=0 =  (Cortex-A55)
inside if i=0, LastCore= (Cortex-A55)
This Core i=1 =  (Cortex-A55)
This Core i=2 =  (Cortex-A55)
This Core i=3 =  (Cortex-A55)
This Core i=4 =  (Cortex-A78)
inside if i=4, LastCore= (Cortex-A78)
This Core i=5 =  (Cortex-A78)
This Core i=6 =  (Cortex-A78)
This Core i=7 =  (Cortex-A78)
```

Inside the `BasicSetup()` function when the function is called;
```
        ClusterConfigByCoreType=($(GetCoreClusters))
        ClusterConfig=($(GetRelevantCPUClusters))
```

We get a result of;
```
ClusterConfigByCoreType= 0 4
ClusterConfig= 0 4
```

So there are only two types of core.  However the three policies split them up as;
```
$ cat /sys/devices/system/cpu/cpufreq/policy0/related_cpus
0 1 2 3
$ cat /sys/devices/system/cpu/cpufreq/policy4/related_cpus
4 5 6
$ cat /sys/devices/system/cpu/cpufreq/policy7/related_cpus
7
```

swdee, 2025-05-01:
In GetCPUClusters() the if condition below matches so it returns the `echo "0"` for `${ClustersByCpufreqOrPackageIDs[@]}`.
```
if [ "X${VirtWhat}" != "X" ] && [ "X${VirtWhat}" != "Xnone" ]; then
```
This condition matches because `${VirtWhat}` equals `vm-other`.

In `BasicSetup()` the `VirtWhat` variable is set with;
```
VirtWhat="$(systemd-detect-virt 2>/dev/null)"
```

On the Rubik Pi this call returns;
```
$ systemd-detect-virt  
vm-other
```

On my Rock 5B it returns;
```
$ systemd-detect-virt
none
```

Running debug on `systemd-detect-virt` shows the `qcom,gunyah-hypervisor` which is unknown to `systemd-detect-virt` so it defaults to outputting `vm-other`.
```
SYSTEMD_LOG_LEVEL=debug systemd-detect-virt
Found container virtualization none.
No virtualization found in DMI vendor table.
Unable to read /sys/firmware/dmi/entries/0-0/raw, using the virtualization information found in DMI vendor table, ignoring: No such file or directory
UML virtualization not found in /proc/cpuinfo.
Virtualization XEN not found, /proc/xen does not exist
No virtualization found in CPUID
Virtualization qcom,gunyah-hypervisor-1.0 found in /proc/device-tree/hypervisor/compatible
This platform does not support /proc/sysinfo
Found VM virtualization vm-other
vm-other
```

This is similar to [the bug](https://github.com/systemd/systemd/issues/28113) when running on Dom0 of a Xen host where `systemd-detect-virt` incorrectly detects it being a virtualized environment.

ThomasKaiser, 2025-05-02:
> This condition matches because `${VirtWhat}` equals `vm-other`.

Good catch since then `[ ${#ClustersByCpufreqOrPackageIDs[@]} -lt ${#ClusterConfigByCoreType[@]} ]` matches and `${#ClusterConfigByCoreType[@]}` will be returned!

I hope the new condition fixes it now (potentially overbenchmarking on other systems that got detected as `vm-other`. Thank you again for all your testing/help!

commit b044e188 (2025-05-02): Adding results for Qualcomm QCS6490
Try to cope with qcom,gunyah-hypervisor platforms, minor fixes, closes #116)
closed 2025-05-02 by ThomasKaiser