ThomasKaiser/sbc-bench #117: Is it worth to prefer 64k memory pages for rk3588 boards?

muravjov, 2025-05-21:
Hi, I've built the kernel with CONFIG_ARM64_64K_PAGES=y with upstream v6.14.5 tag for my Radxa ROCK 5 ITX 32GB RAM
and got better results e.g. for 7-zip single,
3128 vs 2585
=> it is 21% faster

See the table for convenience:

```
| Device/details   | Clockspeed |   Kernel |             Distro                   |7-zip multi|7-zip single| AES     | memcpy | memset | kH/s |
| ArmSoM Sige7     | ~2250      |      6.1 | Ubuntu 24.04.1 LTS (noble) arm64     | 16100     | 2551       | 1287270 | 12270  | 30020 |     - |
1|Radxa ROCK 5 ITX | 2304/1800 MHz |   6.1 | Debian GNU/Linux 12 (bookworm) arm64 | 15600     | 2595       | 1316810 | 12900  | 28260 |     - |
2|Radxa ROCK 5 ITX | 2304/1800 MHz | 6.1   | Ubuntu 24.04.1 LTS (noble) arm64     | 15680     | 2585       | 1316750 | 13030  | 28210 | - |
4|Radxa ROCK 5 ITX | ~2310          | 6.14 | Ubuntu 24.04.1 LTS (noble) arm64     | 14180     | 3128       | 1312630 | 13360  | 29100 | - |
| Orange Pi 5 Plus | 2352/1800 MHz | 6.1   | Ubuntu 24.04.1 LTS (noble) arm64     | 16350     | 2602       | 1316480 | 12460  | 30260 | - |
3| Orange Pi 5 Plus| 2304/1800 MHz | 6.1   | Ubuntu 24.04.1 LTS (noble) arm64     | 16290     | 2590       | 1318280 | 12560  | 30180 | - |

1 - native radxa b3 image
2 - Joshua Riek' last image
3 - second run for orange pi, to check power usage under load (<2.2A = 11W)
4 - Joshua Riek' last image + upsteam "stock" latest stable kernel, v6.14.5
```

It is a bit slower for 7-zip multi, but to justify it I benchmarked it on an idle linux (under some load).

muravjov, 2025-05-21:
The full config change against stock armbian config:
```diff
$ git diff main.. -- config/kernel/linux-rockchip64-edge.config | cat
diff --git a/config/kernel/linux-rockchip64-edge.config b/config/kernel/linux-rockchip64-edge.config
index a49c47a82..8aa05531f 100644
--- a/config/kernel/linux-rockchip64-edge.config
+++ b/config/kernel/linux-rockchip64-edge.config
@@ -55,6 +55,7 @@ CONFIG_ARM64_ERRATUM_1286807=y
 CONFIG_ARM64_ERRATUM_1542419=y
 CONFIG_ARM64_ERRATUM_2441009=y
 # CONFIG_ARM64_ERRATUM_2645198 is not set
+CONFIG_ARM64_64K_PAGES=y
 CONFIG_ARM64_VA_BITS_48=y
 CONFIG_SCHED_MC=y
 CONFIG_SCHED_CLUSTER=y
@@ -87,6 +88,8 @@ CONFIG_VIRTUALIZATION=y
 CONFIG_KVM=y
 CONFIG_KPROBES=y
 CONFIG_JUMP_LABEL=y
+CONFIG_ARCH_MMAP_RND_BITS=29
+CONFIG_ARCH_MMAP_RND_COMPAT_BITS=11
 CONFIG_MODULES=y
 CONFIG_MODULE_UNLOAD=y
 CONFIG_MODULE_FORCE_UNLOAD=y
@@ -99,9 +102,7 @@ CONFIG_BINFMT_MISC=m
 CONFIG_ZSWAP=y
 CONFIG_ZSWAP_DEFAULT_ON=y
 CONFIG_ZSWAP_COMPRESSOR_DEFAULT_ZSTD=y
-CONFIG_ZSWAP_ZPOOL_DEFAULT_Z3FOLD_DEPRECATED=y
-CONFIG_ZBUD=y
-CONFIG_ZSMALLOC=y
+CONFIG_ZSWAP_ZPOOL_DEFAULT_ZBUD=y
 # CONFIG_COMPAT_BRK is not set
 CONFIG_KSM=y
 CONFIG_TRANSPARENT_HUGEPAGE=y
@@ -188,7 +189,7 @@ CONFIG_NETWORK_PHY_TIMESTAMPING=y
 CONFIG_NETFILTER=y
 CONFIG_BRIDGE_NETFILTER=m
 CONFIG_NETFILTER_NETLINK_HOOK=m
-CONFIG_NF_CONNTRACK=m
+CONFIG_NF_CONNTRACK=y
 CONFIG_NF_CONNTRACK_SECMARK=y
 CONFIG_NF_CONNTRACK_ZONES=y
 CONFIG_NF_CONNTRACK_EVENTS=y
@@ -208,7 +209,7 @@ CONFIG_NF_CT_NETLINK=m
 CONFIG_NF_CT_NETLINK_TIMEOUT=m
 CONFIG_NF_CT_NETLINK_HELPER=m
 CONFIG_NETFILTER_NETLINK_GLUE_CT=y
-CONFIG_NF_TABLES=m
+CONFIG_NF_TABLES=y
 CONFIG_NF_TABLES_INET=y
 CONFIG_NF_TABLES_NETDEV=y
 CONFIG_NFT_NUMGEN=m
@@ -347,12 +348,12 @@ CONFIG_NFT_FIB_IPV4=m
 CONFIG_NF_TABLES_ARP=y
 CONFIG_NF_LOG_ARP=m
 CONFIG_NF_LOG_IPV4=m
-CONFIG_IP_NF_IPTABLES=m
+CONFIG_IP_NF_IPTABLES=y
 CONFIG_IP_NF_MATCH_AH=m
 CONFIG_IP_NF_MATCH_ECN=m
 CONFIG_IP_NF_MATCH_RPFILTER=m
 CONFIG_IP_NF_MATCH_TTL=m
-CONFIG_IP_NF_FILTER=m
+CONFIG_IP_NF_FILTER=y
 CONFIG_IP_NF_TARGET_REJECT=m
 CONFIG_IP_NF_TARGET_SYNPROXY=m
 CONFIG_IP_NF_NAT=m
@@ -363,7 +364,7 @@ CONFIG_IP_NF_MANGLE=m
 CONFIG_IP_NF_TARGET_ECN=m
 CONFIG_IP_NF_TARGET_TTL=m
 CONFIG_IP_NF_RAW=m
-CONFIG_IP_NF_SECURITY=m
+CONFIG_IP_NF_SECURITY=y
 CONFIG_IP_NF_ARPFILTER=m
 CONFIG_IP_NF_ARP_MANGLE=m
 CONFIG_NFT_DUP_IPV6=m
@@ -431,7 +432,7 @@ CONFIG_L2TP_DEBUGFS=m
 CONFIG_L2TP_V3=y
 CONFIG_L2TP_IP=m
 CONFIG_L2TP_ETH=m
-CONFIG_BRIDGE=m
+CONFIG_BRIDGE=y
 CONFIG_BRIDGE_VLAN_FILTERING=y
 CONFIG_NET_DSA=m
 CONFIG_NET_DSA_TAG_OCELOT=m
@@ -672,7 +673,7 @@ CONFIG_MTD_HYPERBUS=m
 CONFIG_OF_CONFIGFS=y
 CONFIG_ZRAM=m
 CONFIG_ZRAM_WRITEBACK=y
-CONFIG_ZRAM_TRACK_ENTRY_ACTIME=y
+CONFIG_ZRAM_MEMORY_TRACKING=y
 CONFIG_ZRAM_MULTI_COMP=y
 CONFIG_BLK_DEV_LOOP=y
 CONFIG_BLK_DEV_DRBD=m
@@ -798,6 +799,7 @@ CONFIG_TUN_VNET_CROSS_LE=y
 CONFIG_VETH=m
 CONFIG_VIRTIO_NET=m
 CONFIG_NLMON=m
+CONFIG_NETKIT=y
 CONFIG_NET_VRF=m
 CONFIG_MHI_NET=m
 CONFIG_ATM_DUMMY=m
@@ -1153,7 +1155,6 @@ CONFIG_MHI_WWAN_MBIM=m
 CONFIG_MTK_T7XX=m
 CONFIG_XEN_NETDEV_FRONTEND=m
 CONFIG_XEN_NETDEV_BACKEND=m
-CONFIG_VMXNET3=m
 CONFIG_NETDEVSIM=m
 CONFIG_INPUT_MATRIXKMAP=y
 CONFIG_INPUT_MOUSEDEV=y
@@ -3415,7 +3416,6 @@ CONFIG_CRYPTO_USER_API_SKCIPHER=m
 CONFIG_CRYPTO_USER_API_RNG=m
 CONFIG_CRYPTO_USER_API_AEAD=m
 CONFIG_CRYPTO_NHPOLY1305_NEON=m
-CONFIG_CRYPTO_CHACHA20_NEON=y
 CONFIG_CRYPTO_GHASH_ARM64_CE=y
 CONFIG_CRYPTO_SHA1_ARM64_CE=y
 CONFIG_CRYPTO_SHA2_ARM64_CE=y

```

ThomasKaiser, 2025-05-21:
> 3128 vs 2585 => it is 21% faster

When I'm looking into [official results list](https://github.com/ThomasKaiser/sbc-bench/blob/master/Results.md) searching for "Rock 5B" I see 3146 7-ZIP MIPS single-threaded as such the 3128 you achieved seem not so special to me and the _real_ question is why all your other scores are that low?

closed 2025-06-05 by ThomasKaiser

ThomasKaiser, 2025-06-06:
Well, if we look at RK3588 scores generated by others and myself (https://github.com/ThomasKaiser/sbc-bench/blob/master/Sorted-Results.md#7-zip-mips-single-threaded) the 3128 generated by you with some kernel version/setting is somewhat to be expected but also on the low side given you were running all the time a little above 2.3 GHz [1].

Scores lower than 3000 are simply an indication of 'benchmarking gone wrong' which would require to look closely into the detailed sbc-bench output to get an idea why (though me not really interested in this since we know that the A76 in RK3588 at 2.3 GHz have to score above 3000 7-ZIP MIPS single-threaded).

So nothing to do on my side :)

[1] Using the OpenSSL scores it is easy to determine real clockspeed when type of core is known. With an A76 we can divide by 569 and have the actual clockspeed ([details](https://github.com/ThomasKaiser/sbc-bench/blob/master/results/ARMv8-Crypto-Extensions.md)

  * 1287270: 2262 MHz
  * 1312630: 2306 MHz
  * 1316480: 2313 MHz
  * 1316750: 2314 MHz
  * 1316810: 2314 MHz
  * 1318280: 2316 MHz