ThomasKaiser/sbc-bench #121: Benchmarking impossible on fresh install of sbc-bench due to inaccessible tinymembench dependency

mdziekon, 2025-07-27:
When doing a fresh install (`wget ...` and all that on a fresh system) of `sbc-bench`, currently it's impossible to start the benchmarking process due to missing dependency of `tinymembench`. As of right now (2025-07-27), it seem that repo has either been put to being private, or has been removed by the author, see https://github.com/nuumio/tinymembench (points to 404 page). This makes it impossible to finish dependencies resolution stage (script asks for Github credentials to try to reach the repo; when passing garbage data it will display the `Temporary Github problem. Not able to download tinymembench. Please try again later.` error), and continue to benchmarking stage.

ThomasKaiser, 2025-07-28:
Can you please replace 'nuumio' in the script with 'rojaster'?

That's the version Jeff Geerling is using (since it contains fixes to compile on macOS I would assume), that's unfortuately missing several patches compared to the more advanced nuumio version (e.g. RISC-V support). But at least it should build on ARM and x86.

I'm hoping for a temporary issue and regret not having forked the repo before :(

mdziekon, 2025-07-28:
@ThomasKaiser I can confirm this is indeed a feasible workaround, thank you for quick response.

But since the underlying problem is still there, I'll leave the issue open for the time being.

mdziekon, 2025-07-30:
@ThomasKaiser while the workaround worked fine "out of the box", it seems the current solution does not, for some reason. On a freshly installed system, and fresh pull of `sbc-bench`, I'm getting the following error:
```
Installing needed tools: apt-get -f -qq -y install build-essential lm-sensors dmidecode git sysstat lshw links p7zip, tinymembench...Not able to build necessary tools. Aborting.
Most probably gcc and make packages are missing.
./sbc-bench.sh: line 1: kill: (2185) - No such process
```

The last line is not important (I did encounter this with the workaround, yet the script was working fine), but the "abort" part is, as it halts the entire process. This happens both on quite recent Armbian builds (May 2025) with both Ubuntu Noble & Debian Bookworm.

ThomasKaiser, 2025-07-31:
My bad, I forgot to delete the object files ([done few minutes ago](https://github.com/ThomasKaiser/tinymembench/commit/24e84ae660e9ea8c1d80aee6c10883a309efed48)) and those were built for x86.

I tried it out myself now (on ARM too) and it works now :)

mdziekon, 2025-07-31:
I can confirm it does work for me as well. Thanks again for quick intervention :)