ThomasKaiser/sbc-bench #124: Failed to set affinity: invalid argument / division by 0 on Qualcomm SBC

cnxsoft, 2025-09-17:
We're testing the Qualcomm-based Tachyon SBC with sbc-bench.sh, and the script outputs various errors and ultimately fails to run till the end:

`Starting to examine hardware/software for review purposes...
WARNING: dmesg output does not contain early boot messages which
help in identifying hardware details.

It is recommended to reboot now and then execute the benchmarks.
Press [ctrl]-[c] to stop or [enter] to continue.

Average load and/or CPU utilization too high (too much background activity). Waiting...

...
sbc-bench v0.9.72
Installing needed tools: apt-get -f -qq -y install sysstat links mmc-utils smartmontools stress-ng p7zip, tinymembench, ramlat, mhz.., cpufetch (can't build cpuminer) Done.
Checking cpufreq OPP...taskset: failed to set pid 26174's affinity: Invalid argument
taskset: failed to set pid 26176's affinity: Invalid argument
taskset: failed to set pid 26198's affinity: Invalid argument
taskset: failed to set pid 26200's affinity: Invalid argument
taskset: failed to set pid 26220's affinity: Invalid argument
...
 Done.
Executing tinymembench. Done.
Executing RAM latency tester...taskset: failed to set pid 28505's affinity: Invalid argument
taskset: failed to set pid 28514's affinity: Invalid argument
 Done.
Executing OpenSSL benchmark. Done.
Executing 7-zip benchmark. Done.
Throttling test: heating up the device, 5 more minutes to wait. Done.
Checking cpufreq OPP again...taskset: failed to set pid 34832's affinity: Invalid argument
taskset: failed to set pid 34834's affinity: Invalid argument
taskset: failed to set pid 34866's affinity: Invalid argument
taskset: failed to set pid 34868's affinity: Invalid argument
 Done (13 minutes elapsed).

./sbc-bench.sh: line 4595: 100 * MeasuredClockspeedStart / 0 : division by 0 (error token is "0 ")
./sbc-bench.sh: line 1: kill: (32094) - No such process`

Since there's a warning about the kernel log, here it is: https://pastebin.cnx-software.com/?62389628a7557c65#DGhGb89XnjJLqD2DQsQaBLB3GCr4URbw7n5QfHBXktiN

ThomasKaiser, 2025-09-17:
Can you please have a look whether `/var/log/sbc-bench.log` exists and if so share it?

As for the kernel ring buffer: this gets spammed with various stuff (lastly `sda`) and as such the early boot messages are missing. Important on some platforms to do SoC guessing but irrelevant with Qualcomm.

The main failure is proper detection of CPU clusters. Can you please post output of `sbc-bench -m` stopped after a few seconds?

ThomasKaiser, 2025-09-17:
The problem is: can't reproduce: https://github.com/ThomasKaiser/sbc-bench/blob/master/results/Kcd5.txt (with v0.9.72 fetched directly from this repo)

This was with QCM6490's sibling QCS6490 w/o the cellular stuff. The more important difference most probably is kernel version. Radxa runs with a pretty vanilla mainline kernel + some patches on top vs. the Tachyon people relying on an horribly outdated 5.4.219 BSP kernel. It's also noteworthy that these SoCs do *not* contain Kryo cores but plain ARM Cortex instead (don't let us fool by marketing staff).

As for the problem can you please provide the output from

    cat /sys/devices/system/cpu/cpu*/topology/physical_package_id

As a side note for anyone dealing with the Dragon Q6A soon {@geerlinguy and @wtarreau come immediately to my mind): the officially provided Ubuntu images contrary to info spread here and there ('Debug Party' seems to happen solely behind closed curtains on crappy Discord) has the following logon credentials: `ubuntu`:`ubuntu` but the password needs to be changed directly after 1st login (which is something I did eventually with display/keyboard attached since it was too annoying not being able to login via SSH w/o hassles)

wtarreau, 2025-09-22:
BTW regarding login credentials, I also found these ones (`ubuntu:ubuntu`) to work on the default provided image. However, they don't work on the newer images that are available in their github repo. After some testing I figured that `radxa:radxa` works on the t2 image.

ThomasKaiser, 2025-09-30:
Since I still can't reproduce the issue and the review is already published I don't think we're going to make any progress here. As such closing. Feel free to open again if needed. 

> After some testing I figured that radxa:radxa works on the t2 image.

Yep, all the images generated by Radxa's `rbuild` system come with radxa:radxa. But the 1st available on the resources download page used ubuntu:ubuntu als login credentials.